ucb-bar/onnxruntime-riscv #7: problem when I run ./build.sh --parallel

zhou-zhiyuan, 2020-05-19:
**Describe the bug**
problem when I run ./build.sh --parallel

**System information**
[37%] running cpp protocol buffer compiler (lite) on ****/onnx-ml.proto
[36%] running cpp protocol buffer compiler (lite) on ****/onnx-operators-ml.proto
/bin/sh :  ../external/protobuf/cmake/protoc-3.11.1.0 : cannot execute binary file : Exec format error
onnx/CMakeFiles/onnx_proto.dir/build.make : 62 recipe for target 'onnx/onnx-ml-pb.h' failed
/bin/sh :  ../external/protobuf/cmake/protoc-3.11.1.0 : cannot execute binary file : Exec format error
make[2] : ***[onnx/onnx-ml.pb.h] Error 126
make[2] : ***Waiting for unfinished jobs....
onnx/CMakeFiles/onnx_proto.dir/build.make : 70 recipe for target 'onnx/onnx-operators-ml-pb.h' failed
make[1] : ***[onnx/CMakeFiles/onnx_proto.dir/all] Error 2
Makefile:140:recipe for target 'all' failed
make : *** [all] Error 2

**Expected behavior**
sucessful

**Additional context**
How can I solve this problem?

pranav-prakash, 2020-05-19:
What platform are you building on? We fetch `protoc` for `linux-x86_64` as seen in build.sh

https://github.com/pranav-prakash/onnxruntime-riscv/blob/34dbafc75241cc86c813fc679a55355d7863fd99/build.sh#L16

which we place in `build/protoc/bin/protoc`. Can you make sure you can execute this natively?

Can you also post the full cmake log?

zhou-zhiyuan, 2020-05-20:
Thanks, I solve this problem with your link of protoc. 
My environment is ubuntu 16.04（Virtual Machine）.
But I meet with another question.
I run ./build.sh --parallel under imagenet_runner but log ends up with warning.
system informetion:
********
warning: Using 'dlopen' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking.
Then I run spike --extension=gemmini pk ort_test -m MobileNetV2.onnx  -i images/cat.jpg  -p caffe -x 1 under imagenet_runner but terminate.
system informetion:
******
terminate called after throwing an instance of 'std::runtime_error'
what ( ) could not open pk  (did you misspell it? If VCS,did you forget +permissive/+permissive-off?)

pranav-prakash, 2020-05-20:
>but log ends up with warning.
You can ignore the warning. It is because we are statically linking and glibc does not like to be statically linked. It will not cause any issue in our case.

>what ( ) could not open pk

This error seems to indicate that spike cannot find the proxy kernel. Can you make sure that your installation of spike is correct (in a separate directory, can you successfully compile – using  `riscv64-unknown-linux-gnu-g++ ` and run a simple program – e.g. "hello world" – using `spike --extension=gemmini pk`).

zhou-zhiyuan, 2020-05-21:
I solve this problem by rebuilding spike+pk.
Now I can run spike to simulate network inference.
Next I will use onnxmltools to convert network.
Thanks for your help.

pranav-prakash, 2020-05-21:
Sounds good. Do note that you have to quantize your network in order for it to use Gemmini for the matmul/convolution. You can use the quantization tool in `systolic_runner` folder for this.

closed 2020-05-21 by pranav-prakash