ucb-bar/onnxruntime-riscv #8: Can I generate waveforms from model.onnx

zhou-zhiyuan, 2020-05-22:
**Is your feature request related to a problem? Please describe.**
I can generate waveforms in chipyard by running the command (for example)
./simulator-chipyard-GemminiRocketConfig-debug mobilenet-baremetal .
So can I have some method to use model.onnx to generate waveforms in gemmini?
I want to analyse performance of gemmini and make comparation to other accerelators.

pranav-prakash, 2020-05-22:
Yes you can generate a waveform, but since we can only run in a full linux environment you can't do this via VCS. Instead you'll have to build a firesim fpga image and run that. Assuming you already have firesim set up (follow the docs for firesim if not), you can then build a firemarshal image (you will also have to patch two files in onnxruntime – look at the `BUILD.md` for info on that), and run that on the FPGA. You can then connect to the fpga instance with xilinx logic analyzer, as described [here](https://docs.fires.im/en/latest/Advanced-Usage/Debugging-and-Profiling-on-FPGA/Debugging-Hardware-Using-ILA.html)

@hngenc You might be able to answer more clearly since I don't know what exactly "simulator-chipyard-GemminiRocketConfig-debug" does (I'm guessing it runs it in VCS?). I also don't know if any specific triggers need to be set in order to capture the region of interest.

hngenc, 2020-05-24:
`./simulator-chipyard-GemminiRocketConfig-debug mobilenet-baremetal` will run mobilenet-baremetal on VCS, which will be far too slow to be useful for you. The tests in `imagenet/` are only meant to be run on FireSim, as @pranav-prakash described.

If you use FireSim, then you will only be able to collect a limited number of signals for a limited number of cycles.

It would also help if you could describe which signals exactly you would like to capture from your run. Then, we might be able to give advice on how to collect them.

zhou-zhiyuan, 2020-05-25:
Thanks.I will take look at FireSim snd do more research on gemmini.

closed 2020-05-25 by zhou-zhiyuan