ucb-bar/onnxruntime-riscv #9: compare running time of the same network ( spike  --extension=gemmini ***.onnx vs spike --extension=gemmini***-baremel )

zhou-zhiyuan, 2020-05-29:
I successfully use onnxmltools to convert network and quantize network and use Gemmini for the matmul/convolution.
Now I decide to run gemmini on fgpa (with a risc-v processor). 
Our team will use asic if the performance meets expectations.
You said the mobilenet and resnet50 examples in gemmini-rocc-tests were manually crafted after performing post-training quantization on the two networks.
I want to know running time of the same network (spike ***.onnx  vs spike ***-baremel )
Because I am afraid running .onnx will cost to much time on cpu and cause the gemmini acceleration effect is not obvious .
 And  I don't want to write manually .c and .h file from different pre-trained network. 
I hope everyone can use gemmini automatically as long as he has a pre-trained model on the computer.

pranav-prakash, 2020-05-29:
I have attached trace files that show the timing breakdown collected by running various onnx models via onnxruntime on FPGA – this might be helpful for your needs. You can view them by loading the .json in `chrome://tracing`

[ort0_resnet.zip](https://github.com/pranav-prakash/onnxruntime-riscv/files/4699612/ort0_resnet.zip)
[ort0.zip](https://github.com/pranav-prakash/onnxruntime-riscv/files/4699613/ort0.zip)

I don't have at hand the numbers from running the gemmini-rocc-tests model on FireSim. @hngenc should have those though.

@hngenc Do you happen to know

hngenc, 2020-05-29:
@zhou-zhiyuan Just to be clear: Spike will **not** give you accurate performance numbers. Spike is a functional simulator. It will tell you if your software is written correctly, but it won't tell you how fast or slow that software is.

To get accurate performance numbers, you should use FireSim, which will give you a cycle-accurate simulation.

We've recently updated Gemmini's `master` branch, and rerun the `imagenet/` tests on FireSim. These tests all run one inference with a batch size of 4.

With the latest iteration of Gemmini, using a 16-by-16 systolic array, we achieved the following performance:

`resnet50-linux ws conv`: 824605368 cycles
`mobilenet-linux ws conv`: 4380755903 cycles

Note that in MobileNet, we do depthwise convolutions on the CPU, which _really_ limits performance right now. We intend to port them to run on Gemmini later.

One configuration can achieve different performance from another, so it's also best to run on FireSim yourself to be sure of what your specific configuration's performance will be.

zhou-zhiyuan, 2020-06-02:
I  want to run gemmini at 700M clock. I calculated resnet50, which can run 3 frames per second if it is in baremetal format. There is still a certain gap between onnx and baremetal. 
So I take look at ***.c and ***params.h files. I can read ***.c file (functions are defined) myself.But I don't know how to write ***params.h. Do you have any idea on it? How do I sort out these data from the quantitative onnx model？

hngenc, 2020-06-03:
@zhou-zhiyuan Do you mean the *_params.h files in `imagenet/`, such as `imagenet/resnet50_params.h`?

If so, those were generated using Python scripts we wrote ourselves. The scripts are really hacky right now, and are not that extensible, especially since we're planning to use ONNX-Runtime going forward.

Basically, we quantized the neural networks directly in PyTorch ([this lab tutorial](http://www-inst.eecs.berkeley.edu/~ee290-2/sp20/assets/labs/lab1.pdf) can get you started on how to do that). After that, we looped through the modules of our PyTorch module (`for mod in net.modules():`), printing out a C array in the `imagenet/*_params.h` file for every layer's weights.

zhou-zhiyuan, 2020-06-04:
Thanks for your reply. I will take your advice.

closed 2020-06-04 by zhou-zhiyuan