ucb-bar/onnxruntime-riscv #10: Cannot run some network with spike

J-Zenk, 2020-07-16:
**Describe the bug**
When run onnx models here ```https://github.com/pranav-prakash/onnxruntime-riscv/releases/tag/v0.01```, I got ```bad syscall #131!```
I have tried googlenet_quantized.onnx, mobilenet_quantized_optimized.onnx and resnet50_quantized.onnx, only the resnet could run normally.

**System information**
- OS Platform and Distribution (e.g., Linux Ubuntu 16.04): Linux CentOS 7
- ONNX Runtime installed from (source or binary): source
- Python version:Python: 2.7.5 Python3: 3.6.8

**To Reproduce**
- git clone chipyard and esp-tools
- follow the instruction in chipyard to install esp-tools in chipyard and follow the instruction in systolic_runner to patch riscv-pk in esp-tools and install spike & pk in esp-tools
- change PATH to run the patched spike and pk
- git clone this project
- follow the instruction in systolic_runner to build onnx and ort_test
- download googlenet_quantized.onnx
 from ```https://github.com/pranav-prakash/onnxruntime-riscv/releases/tag/v0.01```
- run ```spike --extension=gemmini pk ort_test -m googlenet_quantized.onnx -i images/cat.jpg -p caffe2 -x 1 -O 0```
Here is the result
```
Gemmini extension configured with:
    dim = 16
bbl loader
Loaded runner program
Using systolic in mode 1
Using Onnxruntime C++ API
terminate called after throwing an instance of 'Ort::Exception'
  what():  /home/zenk/onnxruntime-riscv/onnxruntime/core/session/inference_session.cc:245 onnxruntime::InferenceSession::InferenceSession(const onnxruntime::SessionOptions&, const onnxruntime::Environment&, const string&) status.IsOK() was false. Given model could not be parsed while creating inference session. Error message: Protobuf parsing failed.

bad syscall #131!

```
**Expected behavior**
Get some outputs.

**Additional context**
I also have some other questions. I want to get a network to do face verification. And I found ```https://github.com/onnx/models/tree/master/vision/body_analysis/arcface```. But I tried run the onnx file directly, and it gives me the same bad syscall #131. I also tried to quantize this network by running ```python3 calibrate.py --model_path arcfaceresnet100-8.onnx  --dataset_path ./ --output_model_path arcfaceresnet100-8_quantized.onnx --static=True --data_preprocess=mxnet --mode=int8```. It also failed. 
Here is the result
```
Traceback (most recent call last):
  File "calibrate.py", line 379, in <module>
    main()
  File "calibrate.py", line 348, in main
    args.data_preprocess)
  File "calibrate.py", line 289, in load_test_data
    preprocess_method)
  File "calibrate.py", line 261, in load_single_test_data
    'Number of input protobufs does not match expected model inputs')
ValueError: Number of input protobufs does not match expected model inputs
```

Why I cannot run some network, and why I get the error when try to quantize a network?  How to fix them? Thanks for helping.

pranav-prakash, 2020-07-16:
Hm I've never seen that "Error message: Protobuf parsing failed" before. That syscall seems to correspond to `tgkill` so I think that's just the runner trying to kill itself after the failed protobuf parse.

I cloned a fresh version of the repo (on centos7) and couldn't reproduce the issue, which is going to make this a bit harder to debug. (Just as a quick sanity-check, can you run `md5sum googlenet_quantized.onnx`? It should be `e1360471e07e0810ee3696eb44e66c57`).

The only other major difference I can see at the moment is that your version of `pk` / `spike` seems to be a bit newer than what I've been using (in particular mine doesn't print "bbl loader").

Can you try building `spike` via the following:

* Clone https://github.com/ucb-bar/esp-tools (using the default master branch, latest commit should be `dcb6012f7`)

* Apply the syscall patches inside the riscv-pk folder

* Inside riscv-isa-sim, checkout commit `506d83f6a8d` on the master branch

* Build spike and try using that built version of spike to run.

Hopefully this works, if not we can do some more digging (the fact that protobuf parsing fails seems to indicate that it's a pretty fundamental issue unrelated to the gemmini extension).

------

As for the calibration script, it's written to assume that the `--dataset_path` directory contains a bunch of folders labeled `test_data_set_1`, `test_data_set_2`, etc. each of which contains an `input_0.pb`, `input_1.pb`, and so on. I haven't checked what the folder structure of the `arcface` model looks like, but the error seems to indicate that the folder structure might not match what's described above. 

Also do note that the `imagenet_runner` was hard-coded for using fixed 224x224 pngs as input, so you'll probably have to create your own runner for `arcface`, especially since it seems that the model can accept variable-sized images (so you'll have to manually set the input tensor dimensions – you can look at the upstream onnxruntime api docs for more info).

Also last point for the mxnet derived models is that I've found them to be very finnicky and sensitive to quantization. You might want to first get the non-quantized (i.e. floating point) model running first (for which you'll likely need to write your own runner), at which point you can start playing around with quantization. The accuracy for ResNet models is also a bit poor at the moment because we're limited to power-of-2 scale factors.

pranav-prakash, 2020-07-16:
Oh another thing you can try is to run it with qemu. If you download qemu and enable riscv-user space emulation via `./configure --disable-system --target-list=riscv64-linux-user`, you should get a `qemu-riscv64` binary that you can use instead of spike (just substitute `spike --extension=gemmini pk` for `qemu-riscv64`). Of course if you use qemu you'll have to use `-x 0` (don't use gemmini instructions, i.e. emulate them via cpu only) but this should help us debug where the issue lies.

J-Zenk, 2020-07-16:
Thanks a lot. The md5 value does not match. So I re-download the onnx file with a proxy, and it works.
And I messed the structure of the arcface folder so I got such error. Now inside the folder I have a onnx file and 3 folders named  ```test_data_set_0``` , ```test_data_set_1```, ```test_data_set_2```, each folder contains a ```input_0.pb``` and a ```output_0.pb```. I tried to quantize the network again, it show as below. What causes this error?
```
Num cases 3, num inputs for each cases 1
2020-07-16 17:11:44.181980728 [E:onnxruntime:, sequential_executor.cc:281 Execute] Non-zero status code returned while running BatchNormalization node. Name:'bn0' Status Message: Invalid input scale: NumDimensions() != 3
Traceback (most recent call last):
  File "calibrate.py", line 379, in <module>
    main()
  File "calibrate.py", line 358, in main
    inputs, calib_mode)
  File "calibrate.py", line 118, in get_intermediate_outputs
    for j in range(num_input_names)}) for i in range(num_inputs)
  File "calibrate.py", line 118, in <listcomp>
    for j in range(num_input_names)}) for i in range(num_inputs)
  File "/usr/local/lib64/python3.6/site-packages/onnxruntime/capi/session.py", line 111, in run
    return self._sess.run(output_names, input_feed, run_options)
onnxruntime.capi.onnxruntime_pybind11_state.InvalidArgument: [ONNXRuntimeError] : 2 : INVALID_ARGUMENT : Non-zero status code returned while running BatchNormalization node. Name:'bn0' Status Message: Invalid input scale: NumDimensions() != 3
```

pranav-prakash, 2020-07-16:
Seems like this is relevant

https://github.com/onnx/models/issues/242

J-Zenk, 2020-07-17:
I have tried the fix, and it gives me another error. It seems that the problem is caused by the network, too. Because I also tried googlenet in onnx model zoo and succeed. So maybe i will build a new network and convert it to onnx format and then quantize it. If I do so, Is there anything I should pay attention to, so that I can avoid some strange problems like what happens during the arcface's conversion? Thank you very much.

J-Zenk, 2020-07-18:
> What error did it give?

This is the error. It seems that it is not caused by opset verison, because the origin network is opset version 8 and have the same error as the opset version 9 one which is converted by ```convert_to_opset9.py```.
```
Num cases 3, num inputs for each cases 1
{'conv0': (-2.7700576782226562, 1.8512709140777588), '_mulscalar0': (-1.0126389265060425, -0.9754691123962402), 'stage1_unit1_conv1': (-1.5255255699157715, 1.1363569498062134), 'stage1_unit1_bn1': (-2.677807331085205, 2.351346015930176), 'stage1_unit1_conv2': (-1.0140470266342163, 0.8351755142211914), 'stage1_unit1_relu1': (-0.022769352421164513, 1.9310780763626099), 'stage1_unit1_conv1sc': (-0.3902704417705536, 0.36174634099006653), 'relu0': (-0.5688446164131165, 1.088736653327942), 'stage1_unit2_conv1': (-0.9079145789146423, 1.004768967628479), 'stage1_unit2_bn1': (-0.7862108945846558, 0.780624508857727), 'stage1_unit2_conv2': (-1.0263770818710327, 0.8691070079803467), 'stage1_unit2_relu1': (-0.7567529678344727, 1.049529790878296), 'stage1_unit3_conv1': (-0.5032097697257996, 0.7926806211471558), 'stage1_unit3_bn1': (-1.4762139320373535, 1.992246389389038), 'stage1_unit3_conv2': (-0.4169164001941681, 0.9362981915473938), 'stage1_unit3_relu1': (3.6227927324716802e-09, 2.7148284912109375), 'stage2_unit1_conv1': (-0.580380916595459, 1.6314289569854736), 'stage2_unit1_bn1': (-1.9812438488006592, 4.54368257522583), 'stage2_unit1_conv2': (-1.1352336406707764, 1.2988908290863037), 'stage2_unit1_relu1': (-0.011583560146391392, 5.083901882171631), 'stage2_unit1_conv1sc': (-0.9727423191070557, 1.723375916481018), '_plus2': (-1.670774221420288, 10.384163856506348), 'stage2_unit2_conv1': (-0.2573625147342682, 0.13485988974571228), 'stage2_unit2_bn1': (-0.6998764872550964, 0.27929526567459106), 'stage2_unit2_conv2': (-0.24395932257175446, 0.2444741427898407), 'stage2_unit2_relu1': (3.544041060621339e-09, 1.6052155494689941), 'stage2_unit3_conv1': (-0.33123549818992615, 0.41484981775283813), 'stage2_unit3_bn1': (-0.8922042846679688, 0.544468343257904), 'stage2_unit3_conv2': (-0.5210393071174622, 0.2614585757255554), 'stage2_unit3_relu1': (-0.004366954322904348, 1.0961897373199463), 'stage2_unit4_conv1': (-0.21901430189609528, 0.6279692053794861), 'stage2_unit4_bn1': (-0.9523730278015137, 0.4922447204589844), 'stage2_unit4_conv2': (-0.24297407269477844, 0.27929195761680603), 'stage2_unit4_relu1': (-0.004497249145060778, 0.7889650464057922), 'stage2_unit5_conv1': (-0.397145539522171, 0.7217621207237244), 'stage2_unit5_bn1': (-0.6942195296287537, 0.5839004516601562), 'stage2_unit5_conv2': (-0.21408230066299438, 0.4311734437942505), 'stage2_unit5_relu1': (-0.026107193902134895, 0.5513373613357544), 'stage2_unit6_conv1': (-0.3651759624481201, 0.6158580780029297), 'stage2_unit6_bn1': (-0.5249855518341064, 0.7201350331306458), 'stage2_unit6_conv2': (-0.26098182797431946, 0.27301323413848877), 'stage2_unit6_relu1': (-0.0027470411732792854, 0.5985389947891235), 'stage2_unit7_conv1': (-0.27477845549583435, 0.5691264271736145), 'stage2_unit7_bn1': (-0.6672435998916626, 0.7075446248054504), 'stage2_unit7_conv2': (-0.19847100973129272, 0.2942486107349396), 'stage2_unit7_relu1': (-0.0025474284775555134, 0.44384798407554626), 'stage2_unit8_conv1': (-0.295512318611145, 0.42267584800720215), 'stage2_unit8_bn1': (-0.5995339155197144, 0.6014330387115479), 'stage2_unit8_conv2': (-0.166782408952713, 0.25694501399993896), 'stage2_unit8_relu1': (-0.0017675124108791351, 0.33467692136764526), 'stage2_unit9_conv1': (-0.41656625270843506, 0.37687447667121887), 'stage2_unit9_bn1': (-0.6179551482200623, 0.39605095982551575), 'stage2_unit9_conv2': (-0.2933601438999176, 0.25816017389297485), 'stage2_unit9_relu1': (2.3207581989481696e-07, 0.5152771472930908), 'stage2_unit10_conv1': (-0.4387154281139374, 0.45474547147750854), 'stage2_unit10_bn1': (-0.6467996835708618, 0.5094567537307739), 'stage2_unit10_conv2': (-0.21270819008350372, 0.2974368631839752), 'stage2_unit10_relu1': (2.644956111907959e-07, 0.35188809037208557), 'stage2_unit11_conv1': (-0.35708561539649963, 0.3530902862548828), 'stage2_unit11_bn1': (-0.5718078017234802, 0.4663313329219818), 'stage2_unit11_conv2': (-0.20177653431892395, 0.26275286078453064), 'stage2_unit11_relu1': (9.807602197042797e-08, 0.3139716386795044), 'stage2_unit12_conv1': (-0.2546583414077759, 0.31906643509864807), 'stage2_unit12_bn1': (-0.4647640883922577, 0.44840580224990845), 'stage2_unit12_conv2': (-0.18789871037006378, 0.14361216127872467), 'stage2_unit12_relu1': (9.652539034732399e-08, 0.3494085967540741), 'stage2_unit13_conv1': (-0.41568630933761597, 0.395785927772522), 'stage2_unit13_bn1': (-0.6449421048164368, 0.5393414497375488), 'stage2_unit13_conv2': (-0.3820760250091553, 0.6321218609809875), 'stage2_unit13_relu1': (1.5871721714688647e-08, 0.4499181807041168), 'stage3_unit1_conv1': (-0.5726640820503235, 0.827939510345459), 'stage3_unit1_bn1': (-0.7210996150970459, 0.8041184544563293), 'stage3_unit1_conv2': (-0.4858264923095703, 0.3447681665420532), 'stage3_unit1_relu1': (-0.018802566453814507, 0.411818265914917), 'stage3_unit1_conv1sc': (-1.9470412731170654, 1.312037467956543), '_plus15': (-4.541324615478516, 2.1670632362365723), 'stage3_unit2_conv1': (-0.22424371540546417, 0.14657652378082275), 'stage3_unit2_bn1': (-0.40753498673439026, 0.3977765142917633), 'stage3_unit2_conv2': (-0.17768806219100952, 0.2587220072746277), 'stage3_unit2_relu1': (2.494394664778743e-12, 0.39966338872909546), 'stage3_unit3_conv1': (-0 [... cut ...]
  File "/home/zenk/onnxruntime-riscv/systolic_runner/quantization/quantize.py", line 1296, in _quantize_convolution
    return self._quantize_convolution_qlinear_ops(node, new_nodes_list)
  File "/home/zenk/onnxruntime-riscv/systolic_runner/quantization/quantize.py", line 1188, in _quantize_convolution_qlinear_ops
    self._get_quantization_params(node.output[0])
  File "/home/zenk/onnxruntime-riscv/systolic_runner/quantization/quantize.py", line 644, in _get_quantization_params
    scale_values = [params[1].item()]
AttributeError: 'int' object has no attribute 'item'
```

pranav-prakash, 2020-07-18:
Hm that's very weird. Looks like it's a bug in the quantization script then. Can you post the model you're trying to quantize?

If you'd also like to try debugging this, maybe you could surround
```
      zero_point_values = [params[0].item()]
``` with a try-except and set a pdb breakpoint in the except. That way you can print out the parameters and take a look. It seems weird that `params[1]` exists, but the type isn't what's expected.

pranav-prakash, 2020-07-18:
Ok fixed it! It's a 1 line change :)

https://github.com/pranav-prakash/onnxruntime-riscv/commit/e12f338c469ef09be1c7b8e950aed5202c6fed22

The issue was that the calibration script was returning a non-numpy type in the edge-case where `rmin == rmax` as it was calculating the scale for the quantization parameters. After this fix it successfully saves the quantized model. This bug exists upstream as well. I'll send a PR to them.

Please keep me posted on your results. I'd be interested to see how well the quantized model performs in terms of accuracy. There's a good chance the accuracy might be off at first due to several reasons
* Incorrect preprocessing step during quantization: I think it's possible that the sample data in `input_0.pb`   has already been preprocessed, so you might want to use `--data_preprocess=None` when running quantization script
* Incorrect preprocessing step during inference: They mention some [preprocessing needed](https://github.com/onnx/models/tree/master/vision/body_analysis/arcface#preprocessing) before feeding in the image to mxnet, but it's not clear to me if mxnet itself does further preprocessing of subtracting out the channel mean and dividing by standard deviation as most imagenet models seem to do. You might have to play around with this (you could verify but running the floating point model through a runner you create and seeing if results make sense).
* The quantized model not responding well to rounding scales to power-of-2 factors: If you get to a point where you can run the floating point model through the script successfully but the quantized model behaves weirdly, it's possible that you're running into this issue. What you can try in this case is quantizing to `uint8` instead of `int8` which uses the native `uint8` support in onnxruntime and doesn't suffer from the power-of-2 scale issue. That will quickly tell you if the issue is with the quantization step or with the inference step.

pranav-prakash, 2020-07-20:
>The first error is when running the network with the unmodified runner

`Status Message: Divisor passed to systolic matmul must be power of 2`
That's an error that shouldn't happen since we make sure to round to the nearest power. Can you add a `print` within this function https://github.com/pranav-prakash/onnxruntime-riscv/blob/4c7a0ad5c94fb90bbcc5c876c73e940d3c67d37d/onnxruntime/core/providers/systolic/helper/helper.h to see what the input and output are? 

Also it seems that the network takes an input a `112x112` image that is the result of preprocessing as described [here](https://github.com/onnx/models/tree/master/vision/body_analysis/arcface#preprocessing). You should be able to feed your 224x224 image through the [preprocess script](https://github.com/onnx/models/blob/master/vision/body_analysis/arcface/dependencies/face_preprocess.py ) and then change the runner script to load 112x112 images. I'm not sure why the arcface documentation says "There are no constraints on the size of the image" when the model seems to expect a 112x112 image and the [training data](https://github.com/onnx/models/tree/master/vision/body_analysis/arcface#setup) also used 112x112 images.

As for the `mlperf benchmark`, I have not tried that (that comes from microsoft), and it will likely not work with the gemmini backend anyway.

pranav-prakash, 2020-07-20:
Ok found the issue. It's again an issue with the calibration/quantization script and in the same line as before. In this case `rmin` and `rmax` were both on the order of `1E-41` so the scale was essentially `0` and it overflowed an `int`. Fix is to not compare the floats directly but instead use `np.isclose` which has a default tolerance of `1E-8` which should be fine.

https://github.com/pranav-prakash/onnxruntime-riscv/commit/6475932331f1a0f223221eb40c79959b0f58741f

J-Zenk, 2020-07-22:
Good, now I can run the quantized network. But I cannot run the orign network to get a comparison because of the ```Cannot broadcast gamma to data. gamma: [1,64,1,1], data: [1,64,112,112]``` problem. So I will try to build a new network and quantize it. If there are any problems later, I will open a new issue. Thank you very much for the days help.

closed 2020-07-22 by J-Zenk