ucb-bar/onnxruntime-riscv #12: Supported Gemmini features

pbert519, 2020-08-03:
I'm wondering which features from gemmini v0.3 are currently support by the systolic provider.

pranav-prakash, 2020-08-03:
Currently we only take advantage of Gemmini for quantized convolution (QLinearConv) and quantized matmul (QLinearMatmul) optionally followed by Relu. 

We do not yet support using the hardware im2col feature of Gemmini (I'll likely work on adding this soon), so for now im2col is done by cpu for the convolution. Likewise we do not support hardware-accelerated pooling yet.

>if there is an expected gemmini configuration by the provider, or if all possible variants are working. Always under the assumption, that i modify the systolic_params.h file to fit the current configuration.

All possible variants should work as long as they match the current configuration.

hngenc, 2020-08-04:
v0.3 has `im2col` and pooling on Gemmini. Unfortunately, pooling won't currently work with `Float` types, because we haven't gotten around to implementing the `>` operator for `Float`s yet, but it should work for integer types.

All possible variants should be working, but we usually test with a 16-by-16 systolic array with 8-bit `SInt` inputs and 32-bit accumulated types. If you find that your configuration fails, then please let us know; that's probably a bug.

closed 2020-08-26 by pranav-prakash

pranav-prakash, 2020-09-13:
@pbert519 We've now added support to onnxruntime for scheduling the floating point conv & matmul on Gemmini (assuming you've built it with an fp32 configuration), but the hardware implementation may not be as optimized as it is for int8.

pbert519, 2020-09-14:
Thanks for the update. I will give it a try this week and compare the performance with a int8 version. 

I noticed that the biggest part of the runtime for the quantized models  is used by the QuantizeLinear and im2col when acceleration the multiplication. Gemmini is able to run scaling on move in, is it possible to create a fused operator which combines the quantizeLinear and convolution operations?

pranav-prakash, 2020-09-14:
>give it a try this week 

Follow the instruction in https://github.com/pranav-prakash/onnxruntime-riscv/issues/21#issuecomment-691613525 to enable the floating point supoprt.

>Gemmini is able to run scaling on move in,

We are discussing whether we can implement a more generalized `mvin_scale` that should allow for the float -> int8 conversion and multiplication of scale that you suggest (assuming I'm understanding correctly). We'll get back to you in a week or so when we decide how to proceed on that.

>biggest part of the runtime for the quantized models is used by the QuantizeLinear and im2col when acceleration the multiplication

Yes we saw the same thing on our own benchmarks (trace image attached). Thank you for bringing this up though – I believe Gemmini has recently added support for direct hardware convolution, but we haven't made use of this within onnxruntime yet. We'll also discuss and see if we can make use of this within onnxruntime, which should hopefully remove the need for the `im2col` entirely.

[image]

pranav-prakash, 2020-09-18:
We discussed the quantization as part of `mvin_scale` and decided that the quantizelinear is probably better suited for a vector accelerator (e.g. hwacha) rather than trying to add the float->int8 conversion as part of systolic's mvin. That being said, it might end up being needed for fp16 related stuff, so *if* it is eventually added we can probably extend it to handle the quantization as well.

As for the im2col, since gemmini supports hardware convolution we're planning to try mapping nhwc convs onto gemmini. We have an onnxruntime pass that does the nchw -> nhwc conversion, and in most cases where there are back to back convs there shouldn't be any extra transposes inserted inbetween.