ucb-bar/onnxruntime-riscv #14: error while building onnxruntime using cross-compiilation

harry609, 2020-08-28:
**Describe the bug**
[ 72%] Linking CXX static library libonnxruntime_providers.a
[ 72%] Built target onnxruntime_providers
Scanning dependencies of target onnx_test_runner
Scanning dependencies of target onnxruntime
[ 72%] Building C object CMakeFiles/onnxruntime.dir/generated_source.c.o
Scanning dependencies of target onnxruntime_test_all
[ 72%] Building CXX object CMakeFiles/onnx_test_runner.dir/home/onnxruntime-riscv/onnxruntime/test/onnx/main.cc.o
In file included from /home/onnxruntime-riscv/build/Release/generated_source.c:1:
/home/onnxruntime-riscv/include/onnxruntime/core/providers/systolic/systolic_provider_factory.h:14:137: error: expected ';', ',' or ')' before '=' token
   14 | ORT_API_STATUS(OrtSessionOptionsAppendExecutionProvider_Systolic, _In_ OrtSessionOptions* options, int use_arena, char accelerator_mode = 0)
      |                                                                                                                                         ^
/home/onnxruntime-riscv/include/onnxruntime/core/session/onnxruntime_c_api.h:168:76: note: in definition of macro 'ORT_API_STATUS'
  168 |   ORT_EXPORT _Check_return_ _Ret_maybenull_ OrtStatusPtr ORT_API_CALL NAME(__VA_ARGS__) NO_EXCEPTION ORT_MUST_USE_RESULT
      |                                                                            ^~~~~~~~~~~
/home/onnxruntime-riscv/build/Release/generated_source.c: In function 'GetFunctionEntryByName':
/home/onnxruntime-riscv/build/Release/generated_source.c:7:89: error: 'OrtSessionOptionsAppendExecutionProvider_Systolic' undeclared (first use in this function); did you mean 'OrtSessionOptionsAppendExecutionProvider_CPU'?
    7 | if(strcmp(name,"OrtSessionOptionsAppendExecutionProvider_Systolic") ==0) return (void*)&OrtSessionOptionsAppendExecutionProvider_Systolic;
      |                                                                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                                         OrtSessionOptionsAppendExecutionProvider_CPU
/home/onnxruntime-riscv/build/Release/generated_source.c:7:89: note: each undeclared identifier is reported only once for each function it appears in
CMakeFiles/onnxruntime.dir/build.make:90: recipe for target 'CMakeFiles/onnxruntime.dir/generated_source.c.o' failed
make[2]: *** [CMakeFiles/onnxruntime.dir/generated_source.c.o] Error 1
CMakeFiles/Makefile2:724: recipe for target 'CMakeFiles/onnxruntime.dir/all' failed
make[1]: *** [CMakeFiles/onnxruntime.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

**Urgency**
medium

**System information**
- OS Platform and Distribution (e.g., Linux Ubuntu 16.04):Ubuntu 18.04
- ONNX Runtime installed from (source or binary):source
- ONNX Runtime version:1.2.0
- Python version:python-3.6.9
- Visual Studio version (if applicable):
- GCC/Compiler version (if compiling from source):7.5.0
- CUDA/cuDNN version:
- GPU model and memory:

**To Reproduce**
 esp-gnu-toolchain for cross-compilation 
./build.sh --config  Release --build_shared_lib --parallel
**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here. If the issue is about a particular model, please share the model details as well to facilitate debugging.

pranav-prakash, 2020-08-28:
Hm I've never used the `--build_shared_lib` option before. Can you try building without that and seeing if it succeeds? From the error message it seems that  `systolic_provider_factory.cc`  isn't being linked in, which could be because the cmake files for the shared library need to be modified separately.

harry609, 2020-08-29:
Can you try building without that and seeing if it succeeds?
yes without --build_shared_lib option its working fine..

Also is 7.5.0 the version of riscv64-unknown-linux-gnu-g++?
sry i'm using the latest version 9.2.0 ,previously i have updated 7.5.0 by mistake. 

another question i have regarding this onnxruntime .How can we able to produce the import onnxruntime command in our platform once we complete the build successfully??

pranav-prakash, 2020-08-29:
>yes without --build_shared_lib option its working fine..

Great, glad to hear that.

>How can we able to produce the import onnxruntime command in our platform

You will have to use the C/C++ api when interfacing with onnxruntime (while it might technically be possible to build the python module as well, I have never tried that because you'd have to boot a full fedora image).

You can see an example of how the API is used to run imagenet based models (Alexnet, googlenet, etc.) here: https://github.com/pranav-prakash/onnxruntime-riscv/tree/merging/systolic_runner/imagenet_runner

If you have a need to do more advanced preprocessing/postprocessing of input data, I think it would be easiest to do a hybrid approach where you use python for the preprocessing/postprocessing and then call into your runner written in c/c++.

pranav-prakash, 2020-09-03:
Closing since I assume it's resolved? Feel free to open if you have further questions.

closed 2020-09-03 by pranav-prakash