ucb-bar/onnxruntime-riscv #16: Failed to Build ort_runner with Firesim code added.

Entropy-xcy, 2020-09-04:
**Describe the bug**
When I added the code to "runner.cpp" to build for firesim as suggested in BUILD.md, the build will fail with the following error:
```
src/runner.cpp: In function 'int main(int, char**)':
src/runner.cpp:32:16: error: 'MCL_CURRENT' was not declared in this scope
   32 |   if (mlockall(MCL_CURRENT | MCL_FUTURE) != 0) {
      |                ^~~~~~~~~~~
src/runner.cpp:32:30: error: 'MCL_FUTURE' was not declared in this scope
   32 |   if (mlockall(MCL_CURRENT | MCL_FUTURE) != 0) {
      |                              ^~~~~~~~~~
src/runner.cpp:32:7: error: 'mlockall' was not declared in this scope
   32 |   if (mlockall(MCL_CURRENT | MCL_FUTURE) != 0) {
      |
```

pranav-prakash, 2020-09-04:
Did you `#include <sys/mman.h>`? (My fault for not specifying that clearly).

Entropy-xcy, 2020-09-04:
> Did you `#include <sys/mman.h>`? (My fault for not specifying that clearly).

That works! Thank you!

pranav-prakash, 2020-09-04:
No problem! I've updated the docs to make this more clear.

Feel free to open another issue if you run into any other troubles.