ucb-bar/onnxruntime-riscv #17: Unable to build, getting error:libonnxruntime_providers.a(cpu_execution_provider.cc.o)  -- riscv

kirana1996, 2020-09-08:
I am building onnxruntime in RISCV architecture (U540 target).We getting error described bellow,

Describe the bug
Trying to setup onnxruntime on my os and I'm getting error: undefined reference to `onnxruntime::KernelCreateInfo onnxruntime::BuildKernelCreateInfo<onnxruntime::kCpuExecutionProvider_ConcatFromSequence_kOnnxDomain_ver11>()'

Urgency
Moderate. Needed for testing.

System information

OS Platform and Distribution (e.g., Linux): 
ONNX Runtime installed from (source or binary):Source
ONNX Runtime version: 1.3.0
Python version: 3.7
Visual Studio version (if applicable):
GCC/Compiler version (if compiling from source):riscv64-unknown-linux-gnu-gcc/g++ version 8.3
CUDA/cuDNN version: No
GPU model and memory: No, CPU version

To Reproduce
git clone https://github.com/pranav-prakash/onnxruntime-riscvcd onnxruntime
git submodule update --init --recursive
mkdir build
cd build
cmake ../cmake/ -Donnxruntime_ENABLE_PYTHON=ON -DPYTHON_EXECUTABLE=/usr/bin/python3
make

Expected behavior
Successful build
Error:
/usr/lib/gcc/riscv64-oe-linux/8.3.0/../../../../riscv64-oe-linux/bin/ld: libonnxruntime_providers.a(cpu_execution_provider.cc.o):(.data.rel.ro._ZZN11onnxruntime27RegisterOnnxOperatorKernelsERNS_14KernelRegistryEE14function_table+0x990): undefined reference to `onnxruntime::KernelCreateInfo onnxruntime::BuildKernelCreateInfo<onnxruntime::kCpuExecutionProvider_ConcatFromSequence_kOnnxDomain_ver11>()'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/onnxruntime_perf_test.dir/build.make:191: onnxruntime_perf_test] Error 1
make[1]: *** [CMakeFiles/Makefile2:213: CMakeFiles/onnxruntime_perf_test.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

I am not understanding and Can please help to resolve this...

Thanks,

pranav-prakash, 2020-09-08:
>To Reproduce
>cmake ../cmake/ -Donnxruntime_ENABLE_PYTHON=ON -DPYTHON_EXECUTABLE=/usr/bin/python3
>make

Are you trying to compile this natively on a riscv platform, or are you cross-compiling? If you are cross-compiling you should use the build command described in [BUILD.md](https://github.com/pranav-prakash/onnxruntime-riscv/blob/systolic/systolic_runner/docs/BUILD.md).

kirana1996, 2020-09-08:
I trying to compile this natively.  Ok , After modifying  CC= and CXX=, and then modifying the default protoc download , If build success can i able to get import onnxruntime without any provider issue...

pranav-prakash, 2020-09-08:
It should be noted that I have not tried to build this natively. While it *should* hopefully work, I can't guarantee  it nor can I really debug things on my end (I'm happy to assist though).

It seems like you need the python api built as well, so you should also pass `--build_wheel` to `build.sh`. This will generate a python wheel that you can then build.

As I've added in the readme, another thing to note is that if your goal is floating point inference then one main blocker in terms of performance is that the sgemm kernel is implemented via naive matmul. You should probably switch this out with a proper BLAS implementation (that at least provides tiling and better cache utilization).

closed 2020-09-22 by pranav-prakash