ucb-bar/onnxruntime-riscv #18: Runtime Error when running model with Dynamic input size.

Entropy-xcy, 2020-09-09:
Hi, 

On one of my own model with one of the input size:
`name: "input_ids"
type {
  tensor_type {
    elem_type: 7
    shape {
      dim {
        dim_param: "batch"
      }
      dim {
        dim_param: "sequence"
      }
    }
  }
}`

So the ONNX runtime says that my model input has a negative size:
`Input 0 : name=input_ids, type=7, num_dims=2: [-1, -1, ]`

And, I got the following error at runtime:
`terminate called after throwing an instance of 'Ort::Exception'
  what():  /home/entropy/Development/onnxruntime-riscv/onnxruntime/core/framework/tensor.cc:52 void onnxruntime::Tensor::Init(onnxruntime::MLDataType, const onnxruntime::TensorShape&, void*, onnxruntime::AllocatorPtr, ptrdiff_t) shape.Size() must >=0
Stacktrace:
[0x6926da]
[0x48756]
[0x4083a]
[0x487e8]
[0x40a28]
[0x11a5e]
[0x971258]
[0x12e14]

bad syscall #131!`

Is there a way that I can make the dynamic axes (dynamic size in the inputs and outputs) in my ONNX model to be a fixed value?

Thank you!

pranav-prakash, 2020-09-09:
Yes. This is more of an onnxruntime API question, but the documentation *is* a bit confusing to navigate. when you have a dynamic input size, you have to manually overwrite the `-1` with the actual size. See this example in a runner for an RCNN network that I wrote: https://github.com/pranav-prakash/onnxruntime-riscv/blob/merging/systolic_runner/rcnn_runner/src/runner.cpp#L178

Entropy-xcy, 2020-09-09:
> Yes. This is more of an onnxruntime API question, but the documentation _is_ a bit confusing to navigate. When you have a dynamic input size, you have to manually overwrite the `-1` with the actual size. See this example in a runner for an RCNN network that I wrote: https://github.com/pranav-prakash/onnxruntime-riscv/blob/merging/systolic_runner/rcnn_runner/src/runner.cpp#L178
> 
> You don't need to worry about setting the output size since that will be computed by onnxruntime automatically.

So, basically I should delete the code for checking for input sizes and set the input of the model to be a static size, is that correct? Let me try that.

Thank you!

Entropy-xcy, 2020-09-09:
> So, basically I should delete the code for checking for input sizes and set the input of the model to be a static size, is that correct? Let me try that.
>
> Thank you!

I tried your approach and it works. 

Thank you!

pranav-prakash, 2020-09-09:
>So, basically I should delete the code for checking for input sizes and set the input of the model to be a static size, is that correct?

You can keep the part of the code that prints out the input size (since it is mainly for debug), but yes you must eventually overwrite the `-1` with a static size before creating the input tensor.

>I tried your approach and it works.

Great! I'll close this issue for now then!

closed 2020-09-09 by pranav-prakash