ucb-bar/onnxruntime-riscv #20: Cannot Run Model with multiple inputs nodes

Entropy-xcy, 2020-09-09:
Hi,

I am running a model with multiple input nodes. So I tried:
```
Ort::Value batch_input_tensors[3];
batch_input_tensors[0] = input_ids_tensor;
batch_input_tensors[1] = attention_mask_tensor;
batch_input_tensors[2] = token_type_ids_tensor;

auto output_tensors = session.Run(Ort::RunOptions{nullptr}, input_node_names.data(), batch_input_tensors, 3, output_node_names.data(), 2);
```

However, I got the following error at compilation time:
```
src/bert_runner.cpp: In function 'int main(int, char**)':
src/bert_runner.cpp:187:35: error: no matching function for call to 'Ort::Value::Value()'
  187 |   Ort::Value batch_input_tensors[3];
      |                                   ^
In file included from src/bert_runner.cpp:9:
../..//include/onnxruntime/core/session/onnxruntime_cxx_api.h:276:3: note: candidate: 'Ort::Value::Value(Ort::Value&&)'
  276 |   Value(Value&&) = default;
      |   ^~~~~
../..//include/onnxruntime/core/session/onnxruntime_cxx_api.h:276:3: note:   candidate expects 1 argument, 0 provided
../..//include/onnxruntime/core/session/onnxruntime_cxx_api.h:275:12: note: candidate: 'Ort::Value::Value(OrtValue*)'
  275 |   explicit Value(OrtValue* p) : Base<OrtValue>{p} {}
      |            ^~~~~
../..//include/onnxruntime/core/session/onnxruntime_cxx_api.h:275:12: note:   candidate expects 1 argument, 0 provided
../..//include/onnxruntime/core/session/onnxruntime_cxx_api.h:274:12: note: candidate: 'Ort::Value::Value(std::nullptr_t)'
  274 |   explicit Value(std::nullptr_t) {}
      |            ^~~~~
../..//include/onnxruntime/core/session/onnxruntime_cxx_api.h:274:12: note:   candidate expects 1 argument, 0 provided
src/bert_runner.cpp:188:28: error: use of deleted function 'Ort::Value& Ort::Value::operator=(const Ort::Value&)'
  188 |   batch_input_tensors[0] = input_ids_tensor;
      |                            ^~~~~~~~~~~~~~~~
In file included from src/bert_runner.cpp:9:
../..//include/onnxruntime/core/session/onnxruntime_cxx_api.h:256:8: note: 'Ort::Value& Ort::Value::operator=(const Ort::Value&)' is implicitly declared as deleted because 'Ort::Value' declares a move constructor or move assignment operator
  256 | struct Value : Base<OrtValue> {
      |        ^~~~~
src/bert_runner.cpp:189:28: error: use of deleted function 'Ort::Value& Ort::Value::operator=(const Ort::Value&)'
  189 |   batch_input_tensors[1] = attention_mask_tensor;
      |                            ^~~~~~~~~~~~~~~~~~~~~
src/bert_runner.cpp:190:28: error: use of deleted function 'Ort::Value& Ort::Value::operator=(const Ort::Value&)'
  190 |   batch_input_tensors[2] = token_type_ids_tensor;
      |                            ^~~~~~~~~~~~~~~~~~~~~
```

So, what is the correct way to run a model with multiple input nodes?

Thank you!

pranav-prakash, 2020-09-09:
Try something like

```

  std::vector<Ort::Value> ort_inputs;
  ort_inputs.push_back(std::move(input_ids_tensor));
  ort_inputs.push_back(std::move(attention_mask_tensor));
  ort_inputs.push_back(std::move(token_type_ids_tensor));

auto output_tensors = session.Run(Ort::RunOptions{nullptr}, input_node_names.data(), ort_inputs.data(), ort_inputs.size(), output_node_names.data(), 2);

```

I think the error you are running into is that `Ort::Value` has no default constructor, so `Ort::Value batch_input_tensors[3];` is invalid. You could alternatively try using an initializer list like `Ort::Value batch_input_tensors[3] = {input_ids_tensor, attention_mask_tensor, token_type_ids_tensor}`

Entropy-xcy, 2020-09-09:
> Try something like
>
> ```
>
>   std::vector<Ort::Value> ort_inputs;
>   ort_inputs.push_back(std::move(input_ids_tensor));
>   ort_inputs.push_back(std::move(attention_mask_tensor)); [...]

Thanks for your solution.

However, I encountered an seg fault during runtime.
```
z  0000000000000000 ra 0000000000999278 sp 000000007f7e65d0 gp 00000000010bde78
tp 00000000010cbfa0 t0 0000000000000001 t1 00000000046d6000 t2 00000000000003ff
s0 000000007f7e6430 s1 0000000001033120 a0 0000000080816000 a1 000000007f807000
a2 0000000000000003 a3 0000000000000022 a4 000000007f807001 a5 000000007fbf6748
a6 0000000000000000 a7 00000000000000de s2 000000000466b8b0 s3 000000000006a750
s4 0000000000000000 s5 fffffffffffff000 s6 0000000001034120 s7 0000000000000fff
s8 0000000080816000 s9 000000007f7e6450 sA 0000000001033120 sB 000000010001d000
t3 000000000000000f t4 000000007f7e641f t5 0000000000000001 t6 0000000000000000
pc 0000000000999194 va 0000000080816008 insn       ffffffff sr 800000020005e020
User store segfault @ 0x0000000080816008
```
Do you have any idea what might be the cause?

pranav-prakash, 2020-09-09:
Hm can you run with ` -d 0`? That will print out verbose log messages.

Also are you running a floating point model or a quantized model?

Entropy-xcy, 2020-09-09:
> Hm can you run with ` -d 0`? That will print out verbose log messages.
> 
> Also are you running a floating point model or a quantized model?

`-d 0` gives me the same result.

My model uses int64_t as a datatype. I use the following code to initialize tensors:
```
#define BATCH 1
#define SEQUENCE 64

  size_t input_ids_size = BATCH * SEQUENCE;
  size_t attention_mask_size = BATCH * SEQUENCE;
  size_t token_type_ids_size = BATCH * SEQUENCE;

  int64_t *input_ids_val = new int64_t[input_ids_size];
  int64_t *attention_mask_val = new int64_t[attention_mask_size];
  int64_t *token_type_ids_val = new int64_t[token_type_ids_size];

  input_node_dims[0][0] = BATCH;
  input_node_dims[0][1] = SEQUENCE;
  input_node_dims[1][0] = BATCH;
  input_node_dims[1][1] = SEQUENCE;
  input_node_dims[2][0] = BATCH;
  input_node_dims[2][1] = SEQUENCE;
  
  auto memory_info = Ort::MemoryInfo::CreateCpu(OrtArenaAllocator, OrtMemTypeDefault);
  Ort::Value input_ids_tensor = Ort::Value::CreateTensor<int64_t>(memory_info, input_ids_val, input_ids_size, input_node_dims[0].data(), input_node_dims[0].size());

  Ort::Value attention_mask_tensor = Ort::Value::CreateTensor<int64_t>(memory_info, attention_mask_val, attention_mask_size, input_node_dims[1].data(), input_node_dims[1].size());

  Ort::Value token_type_ids_tensor = Ort::Value::CreateTensor<int64_t>(memory_info, token_type_ids_val, token_type_ids_size, input_node_dims[2].data(), input_node_dims[2].size());
```
The BATCH and SEQUENCE is the sizes of my inputs, do I need to multiply the size by 2 or 4 because I am using int64_t?

Thank you!

pranav-prakash, 2020-09-09:
>The BATCH and SEQUENCE is the sizes of my inputs, do I need to multiply the size by 2 or 4 because I am using int64_t?

The dimensions seem fine to me. If `-d 0` gives the same result then it seems like the segfault is happening even before inference starts. 

It's annoying to debug these kinds of things since `gdb` is not available (well technically you can include it in the buildroot image and run under qemu). So we'll have to resort to using `printf` debugging (be sure to write to stderr to avoid buffering issues). Can you add some `fprintf(stderr, "message")` log statements to try to narrow down whether the segfault is happening during ` Ort::Value::CreateTensor` or during `session.Run`?

Another thing you can do is compile onnxruntime in debug mode (`--release=Debug`) and then use `objdump` and look up the `pc` where the segfault occurs to find the corresponding function.

Entropy-xcy, 2020-09-10:
> > The BATCH and SEQUENCE is the sizes of my inputs, do I need to multiply the size by 2 or 4 because I am using int64_t?
> 
> The dimensions seem fine to me. If `-d 0` gives the same result then it seems like the segfault is happening even before inference starts.
> 
> It's annoying to debug these kinds of things since `gdb` is not available (well technically you can include it in the buildroot image and run under qemu, but that's annoying to get working). So we'll have to resort to using `printf` debugging (be sure to write to stderr to avoid buffering issues). Can you add some `fprintf(stderr, "message")` log statements to try to narrow down whether the segfault is happening during ` Ort::Value::CreateTensor` or during `session.Run`?
> 
> Another thing you can do is compile onnxruntime in debug mode (`--release=Debug`) and then use `objdump` and look up the `pc` where the segfault occurs to find the corresponding function.

Okay! Thanks for your reply, I will try to see if I can find something.

Entropy-xcy, 2020-09-12:
> The dimensions seem fine to me. If `-d 0` gives the same result then it seems like the segfault is happening even before inference starts.
>
> It's annoying to debug these kinds of things since `gdb` [...]

After some debugging, although I haven't got it resolved, I want to share the following observation with you to see if you have some ideas about it.

Basically, I did the following tests:

1. Run my own program using `-d 0`, which gives me an seg fault and I can trace it to be located at `Ort::Session session(env, model_path, session_options);`
2. Run my own program using default `-d 3`, which gives me an seg fault at `auto output_tensors = session.Run` (They are at different location!).
3. Run the provided `ort_test` with `-d 0`, which also gives me a seg fault at `Ort::Session session(env, model_path, session_options);` like 1.
4. Run my own program with `-d 5`. This time, it does not seg fault. But the following error message is printed:
```
terminate called after throwing an instance of 'Ort::Exception'
  what():  Non-zero status code returned while running Gather node. Name:'Gather_13' Status Message: indices element out of data bounds, idx=71822640 must be within the inclusive range [-30522,30521]
bad syscall #131!
```
I added some prints inside primitive gemmini functions, so I should have got something if the actual inference starts. Now, it seems to me that the program got everything needed for the inference, however something incorrect happens at the beginning of the inference.

Do you have any comments on that? Thanks!

pranav-prakash, 2020-09-12:
Hm thank you for sharing.

>Run the provided `ort_test`

By this do you mean that you ran the binary in the `imagenet_runner` with an imagenet model (e.g. Googlenet/Resnet) and saw that segfault? That shouldn't happen (I just tested with my setup), so if this is the case then it seems the issue lies somewhere with your particular environment, which unfortunately makes it even harder to debug. Please let me know if this is the case.

>  Run my own program with `-d 5`

I actually don't think `-d 5` is valid (seems I made a mistake when writing the command line arguments), since the enum for [OrtLoggingLevel](https://github.com/microsoft/onnxruntime/blob/0dad79b495cd9506fc18c5973114009546c5f9e2/include/onnxruntime/core/session/onnxruntime_c_api.h#L124) only goes `0-4`

But either way it seems very strange that changing the logging level affects program execution. For a given logging level (e.g. `-d 0`) the segfault happens deterministically at the same spot right?

>I added some prints inside primitive gemmini functions

You can actually do better and add some print statements in the main dispatch loop itself (the piece of `onnxruntime` that goes through and invokes each operator).

I've attached a patch for `sequential_executor.cc` which will both dump the execution plan as well as print when each operator is invoked. This will also you to determine if it even begins the inference phase.

```
diff --git a/onnxruntime/core/framework/sequential_executor.cc b/onnxruntime/core/framework/sequential_executor.cc
index af79e8c7..95bd59c4 100644
--- a/onnxruntime/core/framework/sequential_executor.cc
+++ b/onnxruntime/core/framework/sequential_executor.cc
@@ -24,7 +24,7 @@
 #include "core/profile/context.h"
 #endif
 
-// #define TRACE_EXECUTION
+#define TRACE_EXECUTION
 
 // Define this symbol to create Concurrency Visualizer markers.
 // See https://docs.microsoft.com/en-us/visualstudio/profiling/concurrency-visualizer-sdk
@@ -163,8 +163,12 @@ Status SequentialExecutor::Execute(const SessionState& session_state, const std:
   std::cout << std::make_pair(&seq_exec_plan, &session_state) << std::endl;
 #endif
 
+  fprintf(stderr, "Finished dumping execution plan\n");
+
   const auto& graph_viewer = session_state.GetGraphViewer();
 
+   fprintf(stderr, "Got graph viewer\n");
+
 #ifdef CONCURRENCY_VISUALIZER
   // need unique name for the series. number of nodes should be good enough for a subgraph
   char series_name[MaxSeriesNameLengthInChars] = "MainGraph";
@@ -200,7 +204,9 @@ Status SequentialExecutor::Execute(const SessionState& session_state, const std:
       continue;
     }
 
+     fprintf(stderr, "Line 207\n");
     const auto& node = *graph_viewer.GetNode(node_exec_plan.node_index);
+     fprintf(stderr, "Line 209\n");
 
 #ifdef CONCURRENCY_VISUALIZER
     series.write_flag(node.Name().c_str());
@@ -279,6 +285,8 @@ Status SequentialExecutor::Execute(const SessionState& session_state, const std:
       return node.Name().empty() ? MakeString(node.OpType(), "_", node_index) : node.Name();
     }();
 
+   fprintf(stderr, "Computing kernel %s %s\n", node.OpType().c_str(), (node.Name().empty() ? MakeString(node.OpType(), "_", node_index) : node.Name()).c_str());
+
     if (is_profiler_enabled) {
       session_state.Profiler().EndTimeAndRecordEvent(profiling::NODE_EVENT,
                                                      node_name_for_profiling + "_fence_before",
 
@@ -323,6 +331,8 @@ Status SequentialExecutor::Execute(const SessionState& session_state, const std:
     if (is_profiler_enabled) {
       // Calculate total output sizes for this operation.
       CalculateTotalOutputSizes(&op_kernel_context, total_output_sizes, node_name_for_profiling);
+      
+      fprintf(stderr, "Executed op kernel node %s\n", node.OpType().c_str());
 
 #if defined(TRACE_EXECUTION)
       // Trace execution step.
```

Going back to the `-d 0` where it segfaults at `Ort::Session session(env, model_path, session_options);`, can you compile onnxruntime in debug mode `./build.sh --config=Debug --parallel` and then recompile your runner binary (same command `./build.sh --config=Debug`) and then do `riscv64-unknown-linux-gnu-objdump -C -l -d ort_test > dumped.txt`. This might be a large file, but you grep through it to find the `pc` corresponding to the segfault and narrow down the line number.

Entropy-xcy, 2020-09-12:
> By this do you mean that you ran the binary in the `imagenet_runner` with an imagenet model (e.g. Googlenet/Resnet) and saw that segfault? That shouldn't happen (I just tested with my setup), so if [...]

Yes, I ran google net with the following code: `spike --extension=gemmini pk ort_test -m ./googlenet_quantized.onnx  -i images/cat.jpg -d 0 -p caffe2 -x 1 -O 0`. This one should be reproducible.

Actually, I have fixed my program. It seems that I did not initialized part of the inputs so that there is some overflow and out of bound issues. 

For me, the `-d` seems problematic too. It seems `-d 5` is displaying more rich details.

> Going back to the -d 0 where it segfaults at Ort::Session session(env, model_path, session_options);, can you compile onnxruntime in debug mode ./build.sh --config=Debug --parallel and then recompile your runner binary (same command ./build.sh --config=Debug). Now when it segfaults you can do riscv64-unknown-linux-gnu-objdump -C -l -d ort_test > dumped.txt and grep through the file to find the pc corresponding to the segfault and narrow down the line number.

I will try that and let you know.

Thank you!

pranav-prakash, 2020-09-12:
>Yes, I ran google net with the following code

Ok that's a more serious issue then since I just tested it and I cannot reproduce it. Some thoughts: can you try using `-x 0` to use CPU-only instead of using Gemmini? Can you also try running it under `qemu` (you can clone and build qemu with `./configure --disable-system --target-list=riscv64-linux-user --prefix=$RISCV` and then use `qemu-riscv64`).

>It seems `-d 5` is displaying more rich details

Does `-d 5` display richer information than `-d 4`? As mentioned the enums indices only go `0` to `4` so I'm not sure why that would happen.

Entropy-xcy, 2020-09-12:
> Ok that's a more serious issue then since I just tested it and I cannot reproduce it. Some thoughts: can you try using `-x 0` to use CPU-only instead of using Gemmini? Can you also try running it under `qemu` (you can clone and build qemu with `./configure --disable-system --target-list=riscv64-linux-user --prefix=$RISCV` and then use `qemu-riscv64`). Also just to eliminate variables can you try using the floating point googlenet model (you can download it from the onnx model zoo) instead of the quantized one?

umm, that is wired. Actually I have reproduced it from building a whole new environment even from a different computer. I will try to do a fresh install of Linux and a fresh build of the toolchain to see if I can still reproduce it. I will let you know the result after I finished so you don't have to waste time on it.

It might be something completely wrong on my toolchain or the environment that causes these series of issues.  Since I don't quite have any experience with `qemu-riscv64`, I will let you know the result I obtained from firesim. 

> Does `-d 5` display richer information than `-d 4`? As mentioned the enums indices only go `0` to `4` so I'm not sure why that would happen.

So after I fixed the out of range problem, the program will run normally under `-d 3`. `-d 4` seems to give me the same result as `-d 5`. 

Thank you so much for helping me!

pranav-prakash, 2020-09-12:
>Actually I have reproduced it from building a whole new environment even from a different computer.
>after I finished so you don't have to waste time on it.

Thanks, let me know how it goes. While I don't think it would be a compiler issue, I strongly suspect it could be a `riscv pk` issue. When you test on firesim please be sure to also try `-x 0` since I'm fairly certain the instruction encoding of gemmini has changed recently and I haven't updated the libraries to use the new instruction encoding.

> Since I don't quite have any experience with `qemu-riscv64`

It's very easy to use for userspace emulation: just clone `https://github.com/qemu/qemu`, configure as mentioned previously then `make && make install`. You can then just invoke it like `qemu-riscv64 ort_test -m ./googlenet_quantized.onnx -i images/cat.jpg -d 0 -p caffe2 -x 0 -O 0`. I recommend this since it's quicker and easier to iterate than having to launch firesim each time.

closed 2020-09-22 by pranav-prakash