ucb-bar/onnxruntime-riscv #21: Low Gemmini Utilization on floating point model

Entropy-xcy, 2020-09-12:
Hi,

An issue I am having is that: when I am running an unquantized model with `float32` all over the places, by looking at the traces, it seems that the systolic array isn't doing anything at all (almost all the computations including the matrix multiplication is performed by the CPU). Will those computations be performed on the Gemmini if I configure the Gemmini to support floating points as the datatype?

The screenshot for the trace is attached below:
[image]

Thanks!

pranav-prakash, 2020-09-12:
Currently we only use gemmini for `int8` quantized convolution and matmul. While gemmini also recently gained support for floating point, we haven't yet implemented support in onnxruntime to use it.

Is floating-point support crucial for your use-case? If so I'll discuss with the rest of the team to see if we should prioritize implementing this. For just `matmul` it should be straightforward.

Otherwise you can try quantizing your network using the quantization script in the `systolic_runner` folder. I noticed that you were running a BERT-type model, and onnxruntime has a custom `Attention` operator with a quantized equivalent, which should be more performant than the default onnx export which doesn't have an attention operator so has to unroll it into the constituent matmul ops. We don't yet schedule the quantized attention on Gemmini, but I was actually planning to implement it this month since we are also working to benchmark BERT models on Gemmini.

Entropy-xcy, 2020-09-12:
> Is floating-point support crucial for your use-case?

I talked to my team and they agree that floating point would be crucial for the accuracy of BERT. I believe `matmul` would be enough for our BERT model. 

It would be great if the floating point support can be implemented!

Anyway, under the current circumstance, I will try to quantize BERT model and see if the accuracy changes.

Also, does the onnxruntime supports Gemmini with configs that is not default? We would like to look at maybe a 64x64 Gemmini with the following configs:
```
object GemminiConfigs {
  import Arithmetic.FloatArithmetic._

  // val defaultConfig = GemminiArrayConfig[SInt, SInt](
  val defaultConfig = GemminiArrayConfig[Float, Float](
    tileRows = 8,
    tileColumns = 8,
    // meshRows = 4,
    // meshColumns = 4,
    meshRows = 4,
    meshColumns = 4,
    ld_queue_length = 8,
    st_queue_length = 2,
    ex_queue_length = 8,
    rob_entries = 16,
    sp_banks = 4,
    acc_banks = 1,
    sp_capacity = CapacityInKilobytes(256),
    shifter_banks = 1, // TODO add separate parameters for left and up shifter banks
    dataflow = Dataflow.BOTH,
    acc_capacity = CapacityInKilobytes(64),
    mem_pipeline = 1,
    dma_maxbytes = 64, // TODO get this from cacheblockbytes
    dma_buswidth = 128, // TODO get this from SystemBusKey
    aligned_to = 1,
    // inputType = SInt(16.W),
    // outputType = SInt(19.W),
    // accType = SInt(32.W),
    inputType = Float(8, 24),
    outputType = Float(8, 24),
    accType = Float(8, 24),
    // mvin_scale_args = Some(MvinScaleArguments((t: SInt, u: SInt) => t * u, 0, SInt(8.W))),
    // mvin_scale_acc_args = Some(MvinScaleArguments((t: SInt, u: SInt) => t * u, 0, SInt(8.W))),
    mvin_scale_args = None,
    mvin_scale_acc_args = None,
    // mvin_scale_args = Some(MvinScaleArguments((t: Float, u: Float) => t * u, 0, Float(8, 24))),
    // mvin_scale_acc_args = Some(MvinScaleArguments((t: Float, u: Float) => t * u, 0, Float(8, 24))),
    mvin_scale_shared = false,
    pe_latency = 0 
  )
}
```

Thank you!

hngenc, 2020-09-12:
Other configs (like other sizes should also work. If they don't, then that's a bug, and you should let us know so we can fix it.

Unfortunately, we simply have not gotten around to tuning the performance of floating-point implementations. I wouldn't be surprised if it's quite inefficient.

[Also, BERT has been quantized into mixed-precision FP16/FP32 by NVIDIA.](https://github.com/NVIDIA/DeepLearningExamples/tree/master/PyTorch/LanguageModeling/BERT) It might be possible to do it all in FP16 without a large accuracy loss.

Also, just out of curiosity, what would be an acceptable degradation in accuracy for your team?

pranav-prakash, 2020-09-12:
>It would be great if the floating point support can be implemented!

Ok I'll discuss with the team whether we should properly support this in onnxruntime and get back to you by the end of the week.

>I will try to quantize BERT model and see if the accuracy changes.

Sounds good – this should be straightforward to do using the [calibrate.py](https://github.com/pranav-prakash/onnxruntime-riscv/blob/merging/systolic_runner/quantization/calibrate.py) script in the `systolic_runner/quantization` folder.

You can invoke it via

```
python3 calibrate.py --model_path floating_model.onnx  --dataset_path dataset_dir --output_model_path model_quantized.onnx --static=True --data_preprocess=None --mode=int8
```

where `dataset_dir` is a directory that contains folders numbered `test_data_set_1`, `test_data_set_2`, etc. (you need at minimum 1, but can have as many as you want)
Each `test_data_set_#` folder will contain the tensor inputs to the network: `input_0.pb`, `input_1.pb`, etc., where the number of input files corresponds to the number of inputs the model has. Since I think your BERT model has 3 inputs, you will have 3 input files and the numbering of the inputs should be the same as you saw in your `bert_runner`.

Each ".pb" is a tensorproto, which you can convert a numpy array to as described [here](https://github.com/onnx/onnx/blob/master/docs/PythonAPIOverview.md#manipulating-tensorproto-and-numpy-array)

Currently the accuracy of quantized models is a bit degraded because Gemmini only supports power of 2 scaling factors, but we have plans to fix that (hopefully within the next month?) by allowing arbitrary floating-point scale multipliers in `int8` mode.

>onnxruntime supports Gemmini with configs that is not default?

Yes we support that, you just have to ensure that [systolic_params.h](https://github.com/pranav-prakash/onnxruntime-riscv/blob/merging/onnxruntime/core/mlas/lib/systolic/systolic_params.h) matches the `gemmini_params.h` file generated during the gemmini build. Right now the `systolic_include.h` and corresponding `systolic_params.h` target an older version Gemmini instruction encoding, but I'll update it soon (within the next week) to target the latest version which should hopefully fix the issue with Googlenet you ran into.

pranav-prakash, 2020-09-13:
@Entropy-xcy I just pushed support for floating point conv and matmul. You can enable it by setting `onnxruntime_SYSTOLIC_FP32` to `ON` and `onnxruntime_SYSTOLIC_INT8` to `OFF`:

https://github.com/pranav-prakash/onnxruntime-riscv/blob/347f64dc123becbdbd09e7acba5a8299e91148cc/cmake/CMakeLists.txt#L62

And updating `systolic_params.h` with the parameters from floating point, for example

```
#ifndef GEMMINI_PARAMS_H
#define GEMMINI_PARAMS_H
#include <stdint.h>
#include <limits.h>
#define DIM 4
#define ADDR_LEN 32
#define BANK_NUM 4
#define BANK_ROWS 4096
#define ACC_ROWS 4096
#define MAX_BYTES 64
#define MAX_BLOCK_LEN (MAX_BYTES/(DIM*4))
#define MAX_BLOCK_LEN_ACC (MAX_BYTES/(DIM*4))
typedef float elem_t;
elem_t elem_t_max = 3.4028235E38;
elem_t elem_t_min = -3.4028235E38;
typedef float acc_t;
typedef double full_t;
#define ELEM_T_IS_FLOAT
#define ELEM_T_EXP_BITS 8
#define ELEM_T_SIG_BITS 24
#define ACC_T_EXP_BITS 8
#define ACC_T_SIG_BITS 24
typedef uint32_t elem_t_bits;
typedef uint32_t acc_t_bits;
#define HAS_MVIN_SCALE
typedef float scale_t;
typedef uint32_t scale_t_bits;
#define HAS_MVIN_ACC_SCALE
typedef float scale_acc_t;
typedef uint32_t scale_acc_t_bits;
#define row_align(blocks) __attribute__((aligned(blocks*DIM*sizeof(elem_t))))
#define row_align_acc(blocks) __attribute__((aligned(blocks*DIM*sizeof(acc_t))))
#define MVIN_SCALE_ONE 1.0
#endif // GEMMINI_PARAMS_H
```

Let me know if there are any issues.

Entropy-xcy, 2020-09-13:
> @Entropy-xcy I just pushed support for floating point conv and matmul. You can enable it by setting `onnxruntime_SYSTOLIC_FP32` to `ON` and `onnxruntime_SYSTOLIC_INT8` to `OFF`:

Thank you! I will let you know once I tested that!

Entropy-xcy, 2020-09-14:
Hi,

The build failed with the following error:

```
/home/entropy/Development/chipyard/esp-tools-install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: libonnxruntime_common.a(env.cc.o): in function `LoadDynamicLibrary':
/home/entropy/Development/gemmini_float_test/onnxruntime-riscv/onnxruntime/core/platform/posix/env.cc:427: warning: Using 'dlopen' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
/home/entropy/Development/chipyard/esp-tools-install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: libonnxruntime_providers_systolic.a(quantize_linear_matmul.cc.o): in function `onnxruntime::systolic::QLinearMatMul<signed char, signed char, signed char>::Compute(onnxruntime::OpKernelContext*) const':
/home/entropy/Development/gemmini_float_test/onnxruntime-riscv/onnxruntime/core/providers/systolic/math/quantize_linear_matmul.cc:84: undefined reference to `SystolicMultiply(char, bool, int, int, int, signed char const*, signed char const*, signed char*, int, float, int const*)'
/home/entropy/Development/chipyard/esp-tools-install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: libonnxruntime_providers_systolic.a(qlinearconv.cc.o): in function `onnxruntime::systolic::QLinearConv<(onnxruntime::StorageOrder)1>::Compute(onnxruntime::OpKernelContext*) const':
/home/entropy/Development/gemmini_float_test/onnxruntime-riscv/onnxruntime/core/providers/systolic/nn/qlinearconv.cc:252: undefined reference to `SystolicMultiply(char, bool, int, int, int, signed char const*, int, signed char const*, int, signed char*, int, int, float, int const*, int, bool)'
/home/entropy/Development/chipyard/esp-tools-install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: libonnxruntime_providers_systolic.a(qlinearconv.cc.o): in function `onnxruntime::systolic::QLinearConv<(onnxruntime::StorageOrder)2>::Compute(onnxruntime::OpKernelContext*) const':
/home/entropy/Development/gemmini_float_test/onnxruntime-riscv/onnxruntime/core/providers/systolic/nn/qlinearconv.cc:502: undefined reference to `SystolicMultiply(char, bool, int, int, int, signed char const*, signed char const*, signed char*, int, float, int const*)'
collect2: error: ld returned 1 exit status
```

pranav-prakash, 2020-09-14:
It should not be compiling `QLinearMatMul` if you have set it to floating point mode. Is this on a clean build (i.e. you deleted the `build//` directory before compiling?)

Entropy-xcy, 2020-09-14:
> Hm it should not be compiling `QLinearMatMul` if you have set it to floating point mode. Is this on a clean build (i.e. you deleted the `build/` directory before compiling?). Also please double check that you set `onnxruntime_SYSTOLIC_INT8` to `OFF`.

I made a mistake when building. This issue is resolved now!

I build my programs running the floating point model again. However, this time, I got some runtime errors:
```
Start Initiating Session!
z  0000000000000000 ra 0000000000a1845a sp 000000007f7e8380 gp 000000000117bd10
tp 000000000118a520 t0 0000000000000000 t1 000000000187b000 t2 0000000000001000
s0 000000007f7e8130 s1 00000000010ec538 a0 0000000080816000 a1 000000007f809000
a2 0000000000000003 a3 0000000000000022 a4 000000007f809001 a5 000000007faf81c0
a6 0000000000000000 a7 00000000000000de s2 00000000017d5d70 s3 00000000000a5290
s4 0000000000000000 s5 fffffffffffff000 s6 00000000010ed538 s7 0000000000000fff
s8 0000000080816000 s9 000000007f7e8150 sA 00000000010ec538 sB 000000010001f000
t3 0000000000000061 t4 000000007f7e811f t5 0000000000000000 t6 0000000000080000
pc 0000000000a18376 va 0000000080816008 insn       ffffffff sr 800000020005e020
User store segfault @ 0x0000000080816008
```
I can trace this to `Ort::Session session(env, model_path, session_options);` in the `runner.cpp`:
```
  printf("Start Initiating Session!\n");
  fflush(stdout);
  Ort::Session session(env, model_path, session_options);
  printf("Session Initiated!\n");
  fflush(stdout);
```

Do you have any ideas about that?

Thank you!

pranav-prakash, 2020-09-14:
Perhaps this related to the segfault you saw before? Are you running this with `spike`? Maybe you can try running it using `qemu` as described in `https://github.com/pranav-prakash/onnxruntime-riscv/issues/20#issuecomment-691397186` to help rule out a spike related issue?

Entropy-xcy, 2020-09-15:
> Perhaps this related to the segfault you saw before? Are you running this with `spike`? Maybe you can try running it using `qemu` userspace-emulation as described in [#20 (comment)](https://github.com/pranav-prakash/onnxruntime-riscv/issues/20#issuecomment-691397186) and [#20 (comment)](https://github.com/pranav-prakash/onnxruntime-riscv/issues/20#issuecomment-691403399) (those link to 2 different comments) to help rule out a spike related issue?

Sorry that I asked so many questions. But where should I put the `gemmini_params.h` file to build a qemu specified for the particular gemmini hardware? I searched around and I can't find an answer.

Entropy-xcy, 2020-09-15:
I ran the original software on qemu (The qemu is not patched with `gemmini_params.h`). It ran without any seg faults. However, it seems the systolic array still isn't been used.
[image]

So I tried to see if the qemu works with the binaries of `gemmini-rocc-tests`, that still does not seem to work, I got the following error while running those binaries on qemu. Seems lacking some libraries. 
```
$ qemu-riscv64 matmul_os-linux
/lib/ld-linux-riscv64-lp64d.so.1: No such file or directory
```

pranav-prakash, 2020-09-15:
>it seems the systolic array still isn't been used.

Do you see the `Using accelerated matmul with dimensions ...` being printed out during inference? Can you run with `-d 0` and paste the output here? Also just to double check (since I sometimes forget to do so myself) you did recompile the runner binary as well right?

>on qemu

Note that qemu does not actually support gemmini (as in if you try to issue the instruction it will fault). However, when you run with `-x 0` it will simulate gemmini (more precisely, this corresponds to the `CPU` `tiled_matmul_type_t` as seen in `gemmini_rocc_tests). 

>So I tried to see if the qemu works with the binaries of `gemmini-rocc-tests`

Binaries that run under qemu userspace need to be statically linked. It seems that `matmul_os-linux` is not. I wouldn't worry about that. 

----

Since it seems the binary runs without segfaulting in `qemu`, I think there might be something weird with the recent version of spike or pk. 

For reference, this is the versions of pk/spike that I have and which works for me:

riscv-pk: 191fc4bfab264b (with the futex syscall patched as mentioned in the readme)
esp-isa-sim: 506d83f6 (with the two patches from 6bd85fd9 and 4fe027c17f added)

I'm fairly certain that spike is well tested, but `pk` may be at fault, so perhaps you could try checking out the latest `esp-isa-sim` and using commit `191fc4bfab264b` of `riscv-pk`.

----

Another option is to build an image in FireMarshal and then run that under `spike`. In this case you'll be building a complete linux image like you do for FireSim, and running that directly under `spike` (without `pk`).

pranav-prakash, 2020-09-15:
Actually wait I can replicate your issue with the bert model I recently pushed. I'll take a look – give me a day or so.

pranav-prakash, 2020-09-15:
I think I fixed the issue. Please pull the latest commit and try that.

pranav-prakash, 2020-09-15:
I was able to replicate your issue of it not using systolic on the bert model I recently pushed.

I think I have found and fixed the issue. Please pull the latest commit and try again.

Entropy-xcy, 2020-09-15:
> I was able to replicate your issue (of it not using systolic) using the bert model runner I recently pushed.
> 
> I think I have found and fixed the issue. Please pull the latest commit and try again, sorry for the inconvenience.

Thank you very much! I will try and let you know. You and your team are great!

Entropy-xcy, 2020-09-15:
> I was able to replicate your issue (of it not using systolic) using the bert model runner I recently pushed.
> 
> I think I have found and fixed the issue. Please pull the latest commit and try again, sorry for the inconvenience.

Hi,

Even though I still haven't successfully got spike and pk to work correctly to run the model (Still seg fault, I guess I will give up on that and use firesim), I do believe that the floating point support is working now. I got the following output while running the program on qemu:
```
Loaded runner program
Using systolic in mode 1
Using Onnxruntime C++ API..
Ready to Run!
Called into systolic matmul!
Using accelerated matmul with dimensions (64, 512, 512)
Illegal instruction
```
Qemu quits on a illegal instruction (because it is a gemmini call) as we expected. That's a good start as it shows that the runtime is assigning the computation of floating point to systolic arrays.

I will try firesim later today and see if it works!

Thank you!

Entropy-xcy, 2020-09-21:
> I will try firesim later today and see if it works!

Hi,

I am very glad to tell you that I have tested on firesim and the floating point implementation works.

If you have nothing to comment, this issue can be closed now.

Thanks for your help!

pranav-prakash, 2020-09-21:
Thank you, that is great to hear!

One thing that might be useful to you is that onnxruntime [supports a fused "Attention" operator](https://wiki.lfai.foundation/download/attachments/22249690/2.3%202020_Apr_ONNX%20MEETUP_ORT%20breakthrough%20optimizations%20for%20transformer%20inference%20on%20GPU%20and%20CPU.pdf?version=1&modificationDate=1586564917000&api=v2).

Since you are interested in BERT models, you might want to take a look at this. I just pushed support for running their quantized Attention (`QAttention`) operator in Gemmini, but you should be able to add support for the floating-point `Attention` operator in the same manner. I'm a bit busy so I can't add and test it myself right now, but if you're interested it should be pretty straightforward, and should accelerate the inference quite a bit.

closed 2020-09-22 by pranav-prakash