ucb-bar/onnxruntime-riscv #27: May I get some help on compilation error?

zhangsenjie-cngdsz, 2021-02-18:
When I was compiling with ./build.sh in top folder (onnxruntime-tiscv), I got this assembler error message for systolic_include.h:
    
    Error: unknown pseudo-op: `.insn'

Could you help me to get the compilation through? 
Thanks a lot.

--------
The onnxruntime-riscv source code is from 2021-01-18 branch and with commit id 85e2cdf76479afbc3b271338d4147e47503780a8.

The esp-tools is from the default master branch and with the latest commit id dcb6012f77101e793948cc90ac31b3735a9f3f6d. 
With this esp-tools, I can run gemmini-rocc-tests successfully. 

--------
Other system information is summarized below (copied from the output of ./build.sh --parallel).

CMake version         : 3.18.0
CMake command         : /usr/local/bin/cmake
System                : Linux
C++ compiler          : /root/Documents/202101/esp-tools-install/bin/riscv64-unknown-linux-gnu-g++
C++ compiler version  : 7.2.0
CXX flags             : -march=rv64imafdc -mabi=lp64d -Wno-error=attributes -Wnon-virtual-dtor
Build type            : Debug
Compile definitions   : ENABLE_ORT_FORMAT_LOAD;EIGEN_MPL2_ONLY;USE_EIGEN_FOR_BLAS
CMAKE_PREFIX_PATH     : 
CMAKE_INSTALL_PREFIX  : /usr/local
CMAKE_MODULE_PATH     : /home/xxxx/Documents/202102/prj_onnx_riscv/onnxruntime-riscv/cmake/external

ONNX version          : 1.8.0
ONNX NAMESPACE        : onnx
ONNX_BUILD_TESTS      : OFF
ONNX_BUILD_BENCHMARKS : OFF
ONNX_USE_LITE_PROTO   : ON
ONNXIFI_DUMMY_BACKEND : OFF
ONNXIFI_ENABLE_EXT    : OFF

Protobuf compiler     : 
Protobuf includes     : 
Protobuf libraries    : 
BUILD_ONNX_PYTHON     : OFF

Looking for strtof_l
Looking for strtof_l - found
Looking for strtoull_l
Looking for strtoull_l - found
Performing Test HAS_UNUSED_BUT_SET_VARIABLE
Performing Test HAS_UNUSED_BUT_SET_VARIABLE - Success
Performing Test HAS_UNUSED_PARAMETER
Performing Test HAS_UNUSED_PARAMETER - Success
Performing Test HAS_UNUSED_VARIABLE
Performing Test HAS_UNUSED_VARIABLE - Success
Performing Test HAS_CAST_FUNCTION_TYPE
Performing Test HAS_CAST_FUNCTION_TYPE - Failed
Performing Test HAS_PARENTHESES
Performing Test HAS_PARENTHESES - Success
Performing Test HAS_USELESS_CAST
Performing Test HAS_USELESS_CAST - Success
Performing Test HAS_NONNULL_COMPARE
Performing Test HAS_NONNULL_COMPARE - Success
Performing Test HAS_TAUTOLOGICAL_POINTER_COMPARE
Performing Test HAS_TAUTOLOGICAL_POINTER_COMPARE - Failed
Performing Test HAS_CATCH_VALUE
Performing Test HAS_CATCH_VALUE - Failed
Performing Test HAS_MISSING_BRACES
Performing Test HAS_MISSING_BRACES - Success
Performing Test HAS_IGNORED_ATTRIBUTES
Performing Test HAS_IGNORED_ATTRIBUTES - Success
Performing Test HAS_DEPRECATED_COPY
Performing Test HAS_DEPRECATED_COPY - Failed
Performing Test HAS_DEPRECATED_DECLARATIONS
Performing Test HAS_DEPRECATED_DECLARATIONS - Success
Performing Test HAS_CLASS_MEMACCESS
Performing Test HAS_CLASS_MEMACCESS - Failed
Performing Test HAS_MAYBE_UNINITIALIZED
Performing Test HAS_MAYBE_UNINITIALIZED - Success
Looking for clock_gettime in rt
Looking for clock_gettime in rt - found

CMake Warning at flake8.cmake:19 (message):
  Could not find 'flake8' to check python scripts.  Please install flake8
  using pip.
Call Stack (most recent call first):
  CMakeLists.txt:1521 (include)

Configuring done
Generating done
CMake Warning:
Manually-specified variables were not used by the project:

onnxruntime_BUILD_WINML_TESTS
onnxruntime_CUDA_HOME
onnxruntime_DNNL_GPU_RUNTIME
onnxruntime_DNNL_OPENCL_ROOT
onnxruntime_MIGRAPHX_HOME
onnxruntime_PYBIND_EXPORT_OPSCHEMA
onnxruntime_ROCM_HOME
onnxruntime_TENSORRT_HOME

Build files have been written to: /home/xxxx/Documents/202102/prj_onnx_riscv/onnxruntime-riscv/build/Debug
2021-02-09 21:33:32,660 util.run [DEBUG] - Subprocess completed. Return code: 0
2021-02-09 21:33:32,660 build [INFO] - Building targets for Debug configuration
2021-02-09 21:33:32,661 util.run [INFO] - Running subprocess in '/home/xxxx/Documents/202102/prj_onnx_riscv/onnxruntime-riscv'
['/usr/local/bin/cmake', '--build', 'build/Debug', '--config', 'Debug', '--', '-j4']

pranav-prakash, 2021-02-18:
Hm `Error: unknown pseudo-op: `.insn'` definitely points to it being a toolchain error. And indeed g++ `C++ compiler version : 7.2.0` is quite out of date (the version I have from installing via Chipyard is gcc 9.2).

If you don't want to install the entire chipyard, can you try using 

https://github.com/ucb-bar/esp-gnu-toolchain

This should be bumped to gcc 9.2

zhangsenjie-cngdsz, 2021-02-21:
I made successful compilation with the esp-tools in chipyard.
And the generated ort_test can be executed on spike+pk.
Thank you very much, Prakash.

However, when I tried to run some example model, like googlenet_quantized.onnx, I got the following message:
`[root@localhost imagenet_runner]# spike --extension=gemmini pk ./ort_test -m googlenet_quantized.onnx -i images/cat.jpg -p caffe2 -x 0 -O 0`
`Gemmini extension configured with:`
`   dim = 16`
`bbl loader`
`Loaded runner program`
`Using systolic in mode 0`
`Using Onnxruntime C++ API`
`terminate called after throwing an instance of 'Ort::Exception'`
`  what():  Could not find an implementation for the node QLinearConv(10)`
`bad syscall #131!`

I guess it is due to the model I used. Is it correct?
If yes, I will make more exploration on model conversion.

pranav-prakash, 2021-02-21:
I think you built the floating point version? See the directions on `build.md` of how to switch it to int8 (Basically toggle the option in cmakelists.txt)

zhangsenjie-cngdsz, 2021-02-23:
Thanks a lot, Prakash.
I modified CMakeLists.txt per your reply. 
Now I can run googlenet_quantized.onnx.

pranav-prakash, 2021-02-23:
Great! I'll go ahead and close the issue then. Let me know iff you run into any other issues!

closed 2021-02-23 by pranav-prakash