ucb-bar/onnxruntime-riscv #29: Assertion failed in gemmini_t::setmode()

zhangsenjie-cngdsz, 2021-02-24:
**Describe the bug**
Using the CPU simulation of Gemmini, I can run the example model successfully as
```
spike --extension=gemmini pk ./ort_test -m googlenet_quantized.onnx -i images/cat.jpg -p caffe2 -x 0 -O 0    
... ...
Element count 1000. Top 5 classes:
0.005367 tiger, Panthera tigris
0.007940 lynx, catamount
0.185520 Egyptian cat
0.186814 tiger cat
0.608402 tabby, tabby cat
Done! Inference took 66829139079 cycles 
```
However, when I use OS mode of Gemmini, there's an assertion failure in gemmini_t::setmode()
```
[root@localhost imagenet_runner]# spike --extension=gemmini pk ./ort_test -m googlenet_quantized.onnx -i images/dog.jpg -p caffe2 -x 1 -O 0
Gemmini extension configured with:
    dim = 16
bbl loader
Loaded runner program
Using systolic in mode 1
Using Onnxruntime C++ API
Number of inputs = 1
Input 0 : name=data_0, type=1, num_dims=4: [1, 3, 224, 224, ]
Number of outputs = 1
Output 0 : name=prob_1, type=1, num_dims=2: [1, 1000, ]
Loading image
Image dimensions: 224 224 3
First few image values 130.061005 126.060997 123.060997
Called into systolic matmul!
Using accelerated matmul with dimensions (64, 12544, 147)
spike: ../gemmini/gemmini.cc:265: void gemmini_t::setmode(reg_t, reg_t): Assertion `new_acc_shift >= 0 && new_acc_shift < sizeof(acc_t)*8' failed.
```

May I have your help in this issue？

**Additional context**
I have checked the consistency between 
```systolic_params_int8.h``` under ```/root/Documents/202102/prj_onnx_riscv/onnxruntime-riscv/onnxruntime/core/mlas/lib/systolic``` 
and 
```gemmini_params.h``` under ```riscv-isa-sim/gemmini```.

pranav-prakash, 2021-02-24:
We don't support `OS` mode well yet. Can you try `WS` (`-x 2`) and see if that works? That should work with `-O 0; there's a known issue where the nhwc mode (`-O 99`) that takes advantage of hardware convolution doesn't work on Spike yet though. @hngenc have you merged the mvin(NULL) fix?

hngenc, 2021-02-24:
@pranav-prakash We are updating `gemmini-rocc-tests` in this PR: https://github.com/ucb-bar/gemmini/pull/69. Unfortunately, we've been getting lots of deadlocks there, which we're trying to resolve. I'll post an update on this issue when we merge that.

@zhangsenjie-cngdsz I think you're probably on the wrong Spike commit. If you run `cat chipyard/generators/gemmini/SPIKE.hash`, then you should see the Spike commit that you should be on. Afterwards, you can run:

```
cd chipyard/toolchains/esp-tools/riscv-isa-sim/build/
git checkout $SPIKE_COMMIT # The output of "cat chipyard/generators/gemmini/SPIKE.hash"
make && make install
```

zhangsenjie-cngdsz, 2021-02-28:
Thank you very much, @pranav-prakash and @hngenc!
It works well with OS and WS mode now.

closed 2021-02-28 by pranav-prakash