ucb-bar/onnxruntime-riscv #40: Not calling Gemmini after fusing QLinearConv and QLinearRelu

zhangsenjie-cngdsz, 2021-06-14:
Hello,

I am using onnxruntime-riscv with branch 2021-05-12 and commit ID 7bbd0496b579863c6906c0449932ac5ddc4c5357.

I found that if operators QLinearConv and QLinearRelu are separated, as the left side of the following figure shown, Gemmini systolic array will be used as expected. However, if they are fused, as the right side of the following figure shown, Gemmini systolic array will NOT be used.  
[image]
Here are the two models: [using_systolic_or_not.zip](https://github.com/ucb-bar/onnxruntime-riscv/files/6645576/using_systolic_or_not.zip)

I don't know this is a desired scheduling, or something wrong?
Could you help?

Thank you very much!

pranav-prakash, 2021-06-14:
Just to double check, you are running models with NHWC transform enabled right (`-O 99`)?

IIRC we don't schedule on gemmini if dimensions for the convolution are non-square or if the number of groups is != 1 (e.g. if it's a depthwise-convolution). You can see the function where we see if we can schedule on Gemmini here:

https://github.com/ucb-bar/onnxruntime-riscv/blob/2021-05-12/onnxruntime/core/providers/systolic/nn/conv_pool_helper.h#L23

```
 * Try to run a given NHWC conv on systolic if possible * Gemmini only supports groups == 1.  * Additionally, we must have strides and padding must be equal * Kernel dimensions must also be square (W == H)
```

zhangsenjie-cngdsz, 2021-06-14:
Thanks a lot, Prakash.
I will check conv_pool_helper.h.

The input 4x1x1024x3 is with NCHW format.
So I didn't enable NHWC transform.
I used -O 1 for the model above.

pranav-prakash, 2021-06-14:
Hm the thing is that if you only used `-O 1` we should have performed the convolution as an im2col on CPU followed by matmul on gemmini (as opposed to `-O 99` which will convert to NHWC format so we can do convolution on gemmini directly) – so with `-O 1` we should _always_ be using Gemmini for the matmul regardless of dimensions.

So even though setting `-O 99` would greatly boost performance and the amount of computation offloaded to Gemmini, I don't think it would explain your case of why the model with fused relu didn't make use of Gemmini at all. 

When you run with `-O 1`, we call into https://github.com/ucb-bar/onnxruntime-riscv/blob/2021-05-12/onnxruntime/core/providers/systolic/nn/qlinearconv.cc#L326 instead which as I mentioned above does an im2col followed by matmul on gemmini: https://github.com/ucb-bar/onnxruntime-riscv/blob/7bbd0496b579863c6906c0449932ac5ddc4c5357/onnxruntime/core/providers/systolic/nn/qlinearconv.cc#L533

Since I've graduated and am no longer the current maintainer of this repo – and the new maintainers are busy with summer internships – we may not be able to look into this ourselves until end of summer. But I think if you try adding some printfs to both the lines mentioned above as well as `conv_pool_helper.h` (depending on whether you are running with `-O 1` or `-O 99`) then that might help reveal why Gemmini isn't being used.

zhangsenjie-cngdsz, 2021-06-24:
Thank you very much for your informative reply.

The model inside file 'not_using_systolic.onnx' contains uint8 initializer for QLinearConv's input 'x_zero_point' and 'y_zero_point', as well as QLinearMatmul's input 'a_zero_point' and 'y_zero_point'. 
So the nodes with op_type QLinearConv or QLinearMatmul were filtered here: 
https://github.com/ucb-bar/onnxruntime-riscv/blob/7bbd0496b579863c6906c0449932ac5ddc4c5357/onnxruntime/core/framework/kernel_registry.cc#L177
This file (not_using_systolic.onnx) was generated by the latest quantize_static method in onnxruntime.quantization package. 
I have set parameter weight_type=QuantType.QInt8. 
I don't know why it generates such uint8/int8 mixed model. 
But your code works in correct behavior definitely.

pranav-prakash, 2021-06-24:
Thank you for the detailed investigation, and we again apologize for not having the time to investigate ourselves.

> latest quantize_static method in onnxruntime.quantization package.

Are you using upstream quantizer package, or our forked version we provide under the `systolic_runner/quantization` directory? Upstream quantizer doesn't fully support int8 yet, perhaps try the one in `systolic_runner/quantization`. If you are still seeing incorrectly generated model with our forked quantizer, then we consider that a bug on our end and will try to fix it.

There is also a pending PR for upstream quantizer to add proper int8 support. Maybe you can try this to see if there is any difference in your generated model: https://github.com/microsoft/onnxruntime/pull/7029#discussion_r657407076

Also by the way, if you are curious about difference between upstream quantizer and our fork, the main difference is explained in [this](https://github.com/ucb-bar/onnxruntime-riscv/blob/2021-05-12/systolic_runner/docs/bumping.md) doc. 

> * Originally the upstream quantizer did not support int8 well, so we had to make a lot of changes to get it working. So it's not as self-contained as the systolic execution provider stuff.

> * The quantizer has had a lot of code churn (it seems like it's partially rewritten every other month), which would have made rebasing annoying. 

We have not updated our quantizer fork in a long time though, and by now most of the int8 related fixes have become available in upstream quantizer as well, but we have not tested it for compatibility with our Gemmini execution provider (for instance, our fork emits a `QLinearRelu` instead of `Relu` because ONNX did not support `int8` relu until very recently in opset 14).

zhangsenjie-cngdsz, 2021-06-25:
The model inside file 'not_using_systolic.onnx' (upper-right figure) is generated by upstream quantizer package. The model corresponding to upper-left figure, which is able to call Gemmini, is generated by the forked version under the systolic_runner/quantization directory.
For my work, the upstream quantizer provides better accuracy using same calibration dataset. It generated a lossless model compared to floating-point model. That's why I am looking at it. 

Thanks a lot again for your information.

pranav-prakash, 2021-06-25:
>For my work, the upstream quantizer provides better accuracy using same calibration dataset. 

Hm could this effect be because the upstream quantizer produced a mixed uint8/int8 model, whereas our forked version produced a purely int8 one? In my experiments I've found that `uint8` quantized models are more accurate than `int8` ones, because `uint8` allows for non-zero values of zero-point but with `int8` you're usually constrained to `zp=0` (onnx model technically supports non-zero zp for int8 as well but I've never seen that actually used anywhere).

But yes you're right that upstream quantizer probably has other improvements, such as calculating scale factors based on histogram entropy instead of just min/max. PR https://github.com/microsoft/onnxruntime/pull/7029#discussion_r657407076 to add official `int8` support in upstream quantizer was recently merged, so maybe you can try that? I think one thing you might have to watch out for is that as I mentioned in https://github.com/microsoft/onnxruntime/pull/7029#issuecomment-800644417 if you're doing `int8` activation then you can't remove the `Relu` after `Conv` like you can with `uint8` activations. Also an annoying thing is that `Relu` with `int8` inputs was only added in opset 14 of ONNX, but we don't yet support that in the Gemmini backend (https://github.com/ucb-bar/onnxruntime-riscv/issues/25. It is a trivial change, but again since the current maintainers are busy with internships over the summer we won't be able to fix it until August. If you do add this though, we'd appreciate any PRs!). If you want to use the upstream quantizer, you can try modifying `operators/activation.py` to emit `QLinearRelu` instead like we do in https://github.com/ucb-bar/onnxruntime-riscv/blob/2021-05-12/systolic_runner/quantization/quantizer/operators/activation.py#L28

Also btw, another approach to producing quantized models in ONNX format is to first perform the quantization in TFLite and then export to ONNX as I briefly describe in https://github.com/ucb-bar/onnxruntime-riscv/blob/2021-05-12/systolic_runner/docs/quantizer.md#exporting-quantized-models. I have not tried this myself, but just wanted to mention it in case it might be useful for you.

zhangsenjie-cngdsz, 2021-06-25:
I am thinking about exactly same thing with you 😄