ucb-bar/onnxruntime-riscv #41: issue of runing onnx model

zhongpanwu, 2021-06-25:
**Describe the bug**
Hello,
after patching the pk and spike, when I run the command "spike --extension=gemmini pk ort_test -m googlenet_quantized.onnx -i images/cat.jpg -p caffe2 -x 1 -O 0", the terminal throws out an error: 

Gemmini extension configured with:
    dim = 16
bbl loader
bad syscall #230!

**System information**
- OS Platform and Distribution:  Linux Ubuntu 18.04
- ONNX Runtime installed from (source or binary):
- ONNX Runtime version: commit 42989847
- Python version: 3.6
- GCC/Compiler version (if compiling from source): riscv64-unknown-elf-g++9.2.0

**To Reproduce**
1, I followed the intructions here: https://github.com/ucb-bar/onnxruntime-riscv/blob/2021-05-12/systolic_runner/docs/BUILD.md.
2, I rebuilt spike and pk separately.
3, I ran the "resnet50-baremetal" from "gemmini-rocc-tests" with the spike, and it worked. Thus I guess there is no issue with spike as it was able to recognize extension of ISA. I suppose the pk has problems, but I did exactly same as the insturctions from the above link. I tried your suggestions like rebuilding the esp_tools, but I got another issue that says "old kerner".

Could you kelp me out, thanks

zhangsenjie-cngdsz, 2021-06-25:
I met same problem before and solved it by updating onnxruntime-riscv to commit 85e2cdf76479afbc3b271338d4147e47503780a8.

pranav-prakash, 2021-06-25:
Yes, please update to latest commit. The submodule given in the gemmini repo is out of date.

zhongpanwu, 2021-06-25:
Thanks for you all replications. 

Now I am using the 2021-05-12 branch, the problem has gone, but another problem popped up now: 

Gemmini extension configured with:
    dim = 16
bbl loader
Loaded runner program
Using systolic in mode 1
Using Onnxruntime C++ API
terminate called after throwing an instance of 'Ort::Exception'
  what():  Could not find an implementation for the node QLinearConv(10)

any clues for this type of error. I tried to set both int8 and fp32 on in CMakeLists.txt file, but problem is still there.

pranav-prakash, 2021-06-25:
In CmakeLists.txt try setting `int8` on, `fp32` off, delete build directory, and rebuild. Deleting build directory is important since otherwise cmake will not pickup changes. You cannot have both `int8` and `fp32` simultaneously on fyi.

zhongpanwu, 2021-06-28:
Sorry, it was not working for me, and showed me the same error.

pranav-prakash, 2021-06-28:
Just to make sure, after rebuilding onnxruntime library you also rebuilt the `ort_test` binary right?

zhongpanwu, 2021-06-30:
Yeah, I also rebuilt ort_test. However, I even faced the problem after I changed  int8 on and fp32 off in the cmake/CMakeList.txt. I was only able to rebuilt ORT successfully with in8 off, fp32 on as default. My cmake version is 3.20.4 in Ubuntu18.04. I am stuckked here for a few days.

pranav-prakash, 2021-06-30:
> I was only able to rebuilt ORT successfully with in8 off, fp32 on as default.

Do you mean that you were getting a _build_ error when you used int8 on, fp32 off? If so could you please post the build log?

Could you also try checking out and using commit https://github.com/ucb-bar/onnxruntime-riscv/commit/85e2cdf76479afbc3b271338d4147e47503780a8 which @zhangsenjie-cngdsz mentioned worked for him? While I don't believe that the newer commits would have broken anything, I had only been testing fp32 so it's possible that something broke when I rebased against upstream.

zhongpanwu, 2021-06-30:
Yes, I got he error during building process in commit 2021-05-12. the log file is attached.
[CMakeError.log](https://github.com/ucb-bar/onnxruntime-riscv/files/6737444/CMakeError.log)

Now I've changed my local repo to the commit  85e2cdf, it also doesn't work for me in building ORT. I notice a new issue complained about missing google/protobuf/inlined_string_field.h. I am sure I have prtobuf 3.15.6 installed.

pranav-prakash, 2021-06-30:
I meant the stdout output from running `./build.sh --parallel`. The messages in `CMakeError.log` are expected I think since the cmake file checks to see if the compiler supports various things

https://github.com/ucb-bar/onnxruntime-riscv/blob/7bbd0496b579863c6906c0449932ac5ddc4c5357/cmake/CMakeLists.txt#L582

Can you please post what is printed to the terminal when you run the  `./build.sh --parallel`?

Also the version of protobuf you have installed should not actually matter, since the `build.sh` script downloads its own version of protobuf.

zhongpanwu, 2021-06-30:
I see, I thought that was the correct log file that you expected. I attached a new file that records the building ORT process.
You may notice one error in line 702: missing function 'StartTime'. I solved this problem by add line "TimePoint StartTime() const" in onnxruntime/core/common/profiler.h. I am not sure if it is correct about this modification, because I still cannot build ORT successfully although without this type of error.  

[build.log](https://github.com/ucb-bar/onnxruntime-riscv/files/6737789/build.log)

pranav-prakash, 2021-06-30:
Thank you for posting, I think this is a bug on our end. When I did the last rebase against upstream, I forgot to update `qlinearconv.cpp` to change `startTime()` to `now()` since I was only testing against fp32.

https://github.com/ucb-bar/onnxruntime-riscv/commit/837b9ec97c3970c6a67b7b6542082df56b17630e#diff-9e2b22e67708c6c2fba9ccbd8b8f6ab63c5b43b5522d48e0380b342045507728

I've asked the current maintainers of this project  (I'm no longer the active maintainer since I've graduated) to take a look but they'll likely only get a chance to work on it in August since they're also busy with summer internships. In the meantime, there are two workarounds you can use to get unstuck:

* You can manually edit the files which are giving build error to change `StartTime()` to `Now()` and change `onnxuntime::make_unique` to `std::make_unique` like in this commit https://github.com/ucb-bar/onnxruntime-riscv/commit/837b9ec97c3970c6a67b7b6542082df56b17630e#diff-9e2b22e67708c6c2fba9ccbd8b8f6ab63c5b43b5522d48e0380b342045507728 – If this works, we'd appreciate if you could send the PR so we can merge it
* Alternatively, you can try out branch `2021-05-09` which is just before the last rebase to upstream. I think the reason why you were getting protobuf build error before was because either the submodule or the protoc version the build script downloaded was incorrect. To get this to work, please make a fresh clone of the repo (the directory doesn't matter, it does not have to be under the chipyard folder so you can clone into the home directory if you like), checkout the `2021-05-09` branch, `git submodule update --init --recursive`, then perform the build.

Once again we apologize for the issue

zhongpanwu, 2021-07-01:
Thank you!  I followed your first workaround and finally I can run the ORT and models on the spike.
Also, I upload my .patch to let you where I changed for fixing bugs.
[patches.txt](https://github.com/ucb-bar/onnxruntime-riscv/files/6745311/patches.txt)

simonguozirui, 2021-12-22:
Thank you all for finding the issue and helping provide a fix. We have fixed this issue in #53, you should be able to run ResNet inference without running to issue related to no implementation for `QLinearConv(10)`. Please use [`2021-12-23`](https://github.com/ucb-bar/onnxruntime-riscv/tree/2021-12-23) branch moving forward. Thanks!

closed 2021-12-22 by simonguozirui