ucb-bar/onnxruntime-riscv #52: Gemmini BERT functionality issue

tntjd7545, 2021-11-21:
**Describe the bug**
BERT inference result of python_inference.py and Gemmini don't match. Gemmini seems to have functionality problem on BERT-base-cased.onnx

**Urgency**
I'd be happy if you could respond until 23rd, November, but I'll understand even if you couldn't

**System information**
- OS Platform and Distribution (e.g., Linux Ubuntu 16.04): Linux Ubuntu 20.04.3 w Docker image from iiswc-gemmini tutorial
- ONNX Runtime installed from (source or binary): pip3 install onnxruntime
- ONNX Runtime version: 1.9.0
- Python version: 3.6
- Visual Studio version (if applicable): none
- GCC/Compiler version (if compiling from source): 7.5.0
- CUDA/cuDNN version: none
- GPU model and memory: none

**To Reproduce**
1. I pulled docker image from iiswc-gemmini tutorial document, and rebuilt everything mentioned in tutorial document
2. Got onnx model following README inside bert_mask_runner. optimize_model.py had nothing to do with the execution result
3. Since I had problem installing onnxruntime inside docker (This is another big issue), I executed python_inference.py at my local machine (system above). I ran python_inference.py on INPUT.py's last sentence and got ['law', 'treaties', 'agreements', 'arbitration', 'relations']
4. Then, I built ort_test and ran gemmini with spike --extension=gemmini pk ort_test -m onnx/bert-base-cased.onnx -x 2 -O 99 after running preprocess.py and got a output.
5. I ran postprocess.py and got garbage value [',', '.', 'the', '##s', '-']

**Expected behavior**
['law', 'treaties', 'agreements', 'arbitration', 'relations']

**Screenshots**
[',', '.', 'the', '##s', '-']

**Additional context**
I also tried INPUT.py's first sentence, but the postprocess.py failed to decode it.
Traceback (most recent call last):
  File "postprocess.py", line 47, in <module>
    print([x["token_str"]  for x in result])
UnicodeEncodeError: 'ascii' codec can't encode character '\uff1a' in position 4: ordinal not in range(128)

tntjd7545, 2021-11-21:
+) I thought the result can be related to quantization issue mentioned in README. I tried to recompile the ort_test using -DPRINT_QUANTIZATION_SCALES flag added to line 45 of /root/chipyard/generators/gemmini/software/onnxruntime-riscv/systolic_runner/bert_mask_runner/build.sh but there was no difference

tntjd7545, 2021-11-22:
+) I used meshRows=meshColumns=8. When running imagenet_runner with tutorial's image and resnet model, I found out that naively changing mesh size results in different inference result. BERT might have the same problem

hngenc, 2021-11-24:
> +) I used meshRows=meshColumns=8. When running imagenet_runner with tutorial's image and resnet model, I found out that naively changing mesh size results in different inference result. BERT might have the same problem

That could be a hardware bug... I will investigate that this week

pranav-prakash, 2021-11-24:
@avinashnandakumar could you take a look to see if you can reproduce this

@tntjd7545 to see if this is a bug in the onnxruntime side or spike model, could you try re-running both the resnet and bert using `-x 0` instead of `-x 2`? This will use cpu-only instead of calling Gemmini.

tntjd7545, 2021-11-24:
> @avinashnandakumar could you take a look to see if you can reproduce this
> 
> @tntjd7545 to see if this is a bug in the onnxruntime side or spike model, could you try re-running both the resnet and bert using `-x 0` instead of `-x 2`? This will use cpu-only instead of calling Gemmini.

@pranav-prakash When I ran it using -x 0, it reporduced the same result as python_inference.py. ['law', 'treaties', 'agreements', 'arbitration', 'relations']

pranav-prakash, 2021-11-24:
@tntjd7545 Thank you, that seems to point at it being a bug in spike. @hngenc and @avinashnandakumar will investigate and get back to you

hngenc, 2021-11-24:
> +) I used meshRows=meshColumns=8. When running imagenet_runner with tutorial's image and resnet model, I found out that naively changing mesh size results in different inference result. BERT might have the same problem

@tntjd7545 I think I know what this issue might be.

Whenever you build Spike using the `./build-spike.sh` script, you generate a new file called `gemmini_params.h`, which you can find in the following location:

```
chipyard/generators/gemmini/software/gemmini-rocc-tests/include/
```

ONNX-Runtime actually has its own copy of this file, called `systolic_params_int8.h`, [found here](https://github.com/ucb-bar/onnxruntime-riscv/blob/2021-05-12/onnxruntime/core/mlas/lib/systolic/systolic_params_int8.h).

Once you changed the meshRows, this file was updated for Spike, but not for ONNX-Runtime.

Can you try replaced `systolic_params_int8.h` with your `gemmini_params.h`, and rebuilding ONNX-Runtime, and seeing if it works then?

This may be unrelated to your other BERT issue, but it might fix the problem of changing meshRows causing incorrect results.

In the future, we will update our scripts so that ONNX-Runtime always uses the same `gemmini_params.h` as Spike does.

tntjd7545, 2021-11-25:
@hngenc  I added the command below to the last line of chipyard/generators/gemmini/script/build-spike.sh
`cp software/gemmini-rocc-tests/include/gemmini_params.h software/onnxruntime-riscv/onnxruntime/core/mlas/lib/systolic/systolic_params_int8.h`

I re-built spike, ran build.sh inside software/onnxruntime-riscv,  and executed spike but spike gave me the same wrong result.

I didn't re-built ort_test because building it make spike fail to show any result. Like the link below:
https://issueexplorer.com/issue/ucb-bar/onnxruntime-riscv/41

tntjd7545, 2021-11-29:
+) I found out that output tensor (shape: [1, 137, 28996]) is composed of 137 identical 28996 dimension tensors including some nan inside.

tntjd7545, 2021-11-29:
+) I found out that the output tensor (shape: [137, 28996]) includes so many NaN and exploded values (e+38). Besides these values, output tensors have almost the same value regardless of their dim[1] (e.g. output[0], output[1], output[2], ... have similar value besides NaN and exploded value)

avinashnandakumar, 2021-12-02:
@tntjd7545 Thanks for your patience, I will look at this issue as soon as I can to try and reproduce it, its currently the last week of classes and finals so there are a few deadlines coming up.

avinashnandakumar, 2021-12-27:
@tntjd7545 Sorry for the late reply, we were finally able to look into this issue. I was able to successfully run the BERT model and get the expected predictions on spike using `-x -0` (CPU) and `-x 2` (Gemmini) but we were using meshRows=meshColumns=16 the default on the onnxruntime branch `2021-12-23` can you confirm that you used this branch? We are testing currently testing meshRows/Columns=8, were you also able to successfully run using dimension size 16?

tntjd7545, 2021-12-27:
@avinashnandakumar I started using gemmini and onnx since your IISWC workshop with the ver2 docker image. So I tried the newest onnxruntime branch today starting again from the docker image, pulled onnxruntime 2021-12-23 branch, ran gemmini/scripts/build-spike.sh and gemmini/software/onnxruntime/build.sh, gemmini/software/onnxruntime/systolic_runner/bert_mask_runner/build.sh. After creating input, I executed spike with "spike --extension=gemmin pk ort_test -m bert-base-cased.onnx -x 2 -O 99"

When I tested with the default config, it resulted in error "LOOP_WS bounds were too large for double-buffering". Therefore, I swtiched to meshRow/Col=8 and got wrong inference results.

If you got the expected result, I think the problem can be related to the docker image...

hngenc, 2021-12-27:
@tntjd7545 It's possible the Docker image is broken. We can look into that.

If you have the time, you can also try installing Gemmini outside of Docker. We've recently provided instructions [here](https://github.com/ucb-bar/gemmini#quick-start) on how to do that. Both Gemmini's `master` and `dev` branches should work.