ucb-bar/onnxruntime-riscv #73: Question: Is there a reason that Gemmini with int8 configuration does not support GEMM operation?

KUEE1080, 2022-03-30:
As the title said, I am just curious if there a specific reason that int8 gemmini does not support GEMM operation compared to fp32 configuration. If there is no specific reason, I am trying to implement this operation in int8 configuration.

hngenc, 2022-03-30:
We do offload int8 GEMMs to Gemmini in fact. We use 8-bit inputs, and accumulate the results into 32-bit partial sums, before scaling the final results back down to 8-bits.

KUEE1080, 2022-03-30:
However, it seems that inside systolic_execution_provider.cc code, only QLinearMatMul, QLinearConv, QLinearConvTranspose, QLinearConv_nhwc is supported. Also, we I run googlenet on spike gemm operation is operated on the CPU. Am I missing something?

<relevent code>
https://github.com/ucb-bar/onnxruntime-riscv/blob/2021-12-23/onnxruntime/core/providers/systolic/systolic_execution_provider.cc#:~:text=of%20op%20kernels-,%23ifdef%20SYSTOLIC_INT8,-class%20ONNX_OPERATOR_TYPED_KERNEL_CLASS_NAME(kSystolicExecutionProvider

<GoogleNet result>
[image]

hngenc, 2022-03-31:
Oh, well, we only run quantized layers on Gemmini. It's possible that the Googlenet implementation keeps the final fully-connected layer in floating point, to maximize model accuracy.

closed 2022-03-31 by KUEE1080