ucb-bar/onnxruntime-riscv #76: Error building the repository

mbelda, 2022-04-26:
Hello,
I am trying to follow the Gemmini ONNX tutorial.
I just run gemmini/scripts/build-onnx-inference.sh that executes the build.sh script on onnxruntime-riscv.
I modified a bit the flags and the one used are the following:  
~~~
./build.sh --config=Release --cmake_extra_defines onnxruntime_USE_SYSTOLIC=ON onnxruntime_SYSTOLIC_INT8=ON onnxruntime_SYSTOLIC_FP32=OFF
~~~

That execution throw an error. The output of the build is next: 

~~~
[ 34%] Linking CXX static library libonnxruntime_graph.a
[ 34%] Built target onnxruntime_graph
Scanning dependencies of target onnxruntime_providers_systolic
[ 34%] Building CXX object CMakeFiles/onnxruntime_providers_systolic.dir/home/user/Documents/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/providers/systolic/fusion/qlinearadd_relu_fuse.cc.o
[ 34%] Building CXX object CMakeFiles/onnxruntime_providers_systolic.dir/home/user/Documents/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/providers/systolic/fusion/qlinearconv_relu_fuse.cc.o
[ 35%] Building CXX object CMakeFiles/onnxruntime_providers_systolic.dir/home/user/Documents/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/providers/systolic/math/gemm.cc.o
[ 35%] Building CXX object CMakeFiles/onnxruntime_providers_systolic.dir/home/user/Documents/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/providers/systolic/math/matmul.cc.o
[ 35%] Building CXX object CMakeFiles/onnxruntime_providers_systolic.dir/home/user/Documents/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/providers/systolic/math/quantize_linear_matmul.cc.o
[ 35%] Building CXX object CMakeFiles/onnxruntime_providers_systolic.dir/home/user/Documents/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/providers/systolic/nn/conv.cc.o
[ 35%] Building CXX object CMakeFiles/onnxruntime_providers_systolic.dir/home/user/Documents/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/providers/systolic/nn/maxpool.cc.o
[ 35%] Building CXX object CMakeFiles/onnxruntime_providers_systolic.dir/home/user/Documents/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/providers/systolic/nn/qlinearconv.cc.o
/home/user/Documents/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/providers/systolic/nn/qlinearconv.cc: In member function 'virtual onnxruntime::common::Status onnxruntime::systolic::QLinearConv_nhwc::Compute(onnxruntime::OpKernelContext*) const':
/home/user/Documents/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/providers/systolic/nn/qlinearconv.cc:218:29: error: 'class onnxruntime::profiling::Profiler' has no member named 'StartTime'; did you mean 'GetStartTimeNs'?
  218 |       start_time = profiler.StartTime();
      |                             ^~~~~~~~~
      |                             GetStartTimeNs
/home/user/Documents/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/providers/systolic/nn/qlinearconv.cc:252:31: error: 'class onnxruntime::profiling::Profiler' has no member named 'StartTime'; did you mean 'GetStartTimeNs'?
  252 |         start_time = profiler.StartTime();
      |                               ^~~~~~~~~
      |                               GetStartTimeNs
/home/user/Documents/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/providers/systolic/nn/qlinearconv.cc:283:31: error: 'class onnxruntime::profiling::Profiler' has no member named 'StartTime'; did you mean 'GetStartTimeNs'?
  283 |         start_time = profiler.StartTime();
      |                               ^~~~~~~~~
      |                               GetStartTimeNs
/home/user/Documents/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/providers/systolic/nn/qlinearconv.cc: In member function 'virtual onnxruntime::common::Status onnxruntime::systolic::QLinearConv::Compute(onnxruntime::OpKernelContext*) const':
/home/user/Documents/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/providers/systolic/nn/qlinearconv.cc:458:31: error: 'class onnxruntime::profiling::Profiler' has no member named 'StartTime'; did you mean 'GetStartTimeNs'?
  458 |         start_time = profiler.StartTime();
      |                               ^~~~~~~~~
      |                               GetStartTimeNs
/home/user/Documents/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/providers/systolic/nn/qlinearconv.cc:502:33: error: 'class onnxruntime::profiling::Profiler' has no member named 'StartTime'; did you mean 'GetStartTimeNs'?
  502 |           start_time = profiler.StartTime();
      |                                 ^~~~~~~~~
      |                                 GetStartTimeNs
/home/user/Documents/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/providers/systolic/nn/qlinearconv.cc:530:31: error: 'class onnxruntime::profiling::Profiler' has no member named 'StartTime'; did you mean 'GetStartTimeNs'?
  530 |         start_time = profiler.StartTime();
      |                               ^~~~~~~~~
      |                               GetStartTimeNs
make[2]: *** [CMakeFiles/onnxruntime_providers_systolic.dir/build.make:154: CMakeFiles/onnxruntime_providers_systolic.dir/home/user/Documents/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/providers/systolic/nn/qlinearconv.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1265: CMakeFiles/onnxruntime_providers_systolic.dir/all] Error 2
make: *** [Makefile:163: all] Error 2
Traceback (most recent call last):
  File "/home/user/Documents/chipyard/generators/gemmini/software/onnxruntime-riscv/tools/ci_build/build.py", line 2160, in <module>
    sys.exit(main())
  File "/home/user/Documents/chipyard/generators/gemmini/software/onnxruntime-riscv/tools/ci_build/build.py", line 2086, in main
    build_targets(args, cmake_path, build_dir, configs, num_parallel_jobs, args.target)
  File "/home/user/Documents/chipyard/generators/gemmini/software/onnxruntime-riscv/tools/ci_build/build.py", line 1087, in build_targets
    run_subprocess(cmd_args, env=env)
  File "/home/user/Documents/chipyard/generators/gemmini/software/onnxruntime-riscv/tools/ci_build/build.py", line 591, in run_subprocess
    return run(*args, cwd=cwd, capture_stdout=capture_stdout, shell=shell, env=my_env)
  File "/home/user/Documents/chipyard/generators/gemmini/software/onnxruntime-riscv/tools/python/util/run.py", line 41, in run
    completed_process = subprocess.run(
  File "/home/user/anaconda3/lib/python3.9/subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/usr/bin/cmake', '--build', 'build/Release', '--config', 'Release']' returned non-zero exit status 2.
Building against Release
riscv64-unknown-linux-gnu-g++: error: ../..//build/Release/libonnxruntime_test_utils.a: No such file or directory
riscv64-unknown-linux-gnu-g++: error: ../..//build/Release/libonnxruntime_session.a: No such file or directory
riscv64-unknown-linux-gnu-g++: error: ../..//build/Release/libonnxruntime_providers.a: No such file or directory
riscv64-unknown-linux-gnu-g++: error: ../..//build/Release/libonnxruntime_util.a: No such file or directory
riscv64-unknown-linux-gnu-g++: error: ../..//build/Release/libonnxruntime_util.a: No such file or directory
riscv64-unknown-linux-gnu-g++: error: ../..//build/Release/libonnxruntime_providers_systolic.a: No such file or directory
riscv64-unknown-linux-gnu-g++: error: ../..//build/Release/libonnxruntime_flatbuffers.a: No such file or directory
riscv64-unknown-linux-gnu-g++: error: ../..//build/Release/external/re2/libre2.a: No such file or directory
riscv64-unknown-linux-gnu-g++: error: ../..//build/Release/external/onnx/libonnx.a: No such file or directory
riscv64-unknown-linux-gnu-g++: error: ../..//build/Release/external/re2/libre2.a: No such file or directory
make: *** [Makefile:11: ort_test] Error 1
Please ignore any dlopen warning above. Glibc hates being statically linked.
~~~

**Urgency**
High.

**System information**
- Linux Ubuntu-20.04
- Python version: 3.9.7
- GCC/Compiler version (if compiling from source): riscv64-unknown-linux-gnu-* from the riscv-tools esp-tools branch
- GPU model and memory:  Gemmini
- Chipyard: 1.6.2 (commit 481398b910fa95ec88dd578c67ba358a4d83129d)
- ONNX-runtime-riscv: commit 7bbd0496b579863c6906c0449932ac5ddc4c5357, origin/2021-05-12
- Gemmini: commit c47cb7f3eb5c18390f176f3a53c43c8546d487d2

**Expected behavior**
Successful installation of the repository.

hngenc, 2022-05-01:
Sorry about that; it seems our subrepos had a version mismatch.

Could you try the latest version of ONNX-Runtime (on branch `2021-12-23`)?

mbelda, 2022-05-04:
With the last version it works fine.

Thank you very much!

hngenc, 2022-05-04:
Thank you for exposing this issue!

closed 2022-05-04 by hngenc