ucb-bar/onnxruntime-riscv #77: Error executing FP32 ONNX model on Gemmini

mbelda, 2022-05-05:
**Describe the bug**
I am trying to execute the model MobilenetV2 a bit modified (just the final layer from 1000 to 13 labels) in FP32 with a the code from systolic_runner/imagenet_runner a little bit modified (dimensions and preprocess).
When I simulate the execution on spike with this command:
~~~
spike --extension=gemmini pk ort_test -m MobileNetV2_0p5_all.onnx -i images/peon_blanco_fondo_blanco.jpg -p mobilenet -x 2 -O 99
~~~

It gives the following error:

~~~
Gemmini extension configured with:
    dim = 16
bbl loader
Loaded runner program
Using systolic in mode 2
Using Onnxruntime C++ API
Number of inputs = 1
Input 0 : name=input, type=1, num_dims=4: [1, 224, 224, 3, ]
Number of outputs = 1
Output 0 : name=dense_2, type=1, num_dims=2: [1, 13, ]
Loading image
Image dimensions: 224 224 3
First few image values -0.929412 -0.803922 -0.780392
Created Ort:CreateCpu
Created Ort:Tensor
Called into systolic matmul!
Using accelerated matmul with dimensions (12544, 16, 27)
LOOP_WS bounds were too large for double-buffering
~~~

I have also tried to run it with the -x 1 option and it gives the following error:
~~~
Gemmini extension configured with:
    dim = 16
bbl loader
Loaded runner program
Using systolic in mode 1
Using Onnxruntime C++ API
Number of inputs = 1
Input 0 : name=input, type=1, num_dims=4: [1, 224, 224, 3, ]
Number of outputs = 1
Output 0 : name=dense_2, type=1, num_dims=2: [1, 13, ]
Loading image
Image dimensions: 224 224 3
First few image values -0.929412 -0.803922 -0.780392
Created Ort:CreateCpu
Created Ort:Tensor
Called into systolic matmul!
Using accelerated matmul with dimensions (12544, 16, 27)
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 1024) >= this->size() (which is 1024)
~~~

**Urgency**
High

**System information**
- OS Platform and Distribution: Linux Ubuntu-20.04
- Python version: 3.9.7
-  GCC/Compiler version (if compiling from source): riscv64-unknown-linux-gnu-* from the riscv-tools esp-tools branch
- GPU model and memory: Gemmini
- Chipyard: 1.6.2 (commit 481398b910fa95ec88dd578c67ba358a4d83129d)
- ONNX-runtime-riscv: commit 0c8c9b4f881b5f31d32c6b5a76cac4ee14a8f338 , 2021-12-23
- Gemmini: commit c47cb7f3eb5c18390f176f3a53c43c8546d487d2

**To Reproduce**
- I attach the model.
[MobileNetV2_0p5_all.zip](https://github.com/ucb-bar/onnxruntime-riscv/files/8632425/MobileNetV2_0p5_all.zip)

- And if you need I can attach the test code.

**Expected behavior**
A correct run, as it runs with -x 0 (CPU).

Thanks in advance!

hngenc, 2022-05-05:
If you're running ONNX-Runtime with FP32 models, then you also need to rebuild Spike with Gemmini's FP config:

https://github.com/ucb-bar/gemmini/blob/master/src/main/scala/gemmini/CustomConfigs.scala#L13

You should be able to rebuild Spike with the `scripts/build-spike.sh` command.

Let me know if that works, or if you need any more clarification

mbelda, 2022-05-05:
I changed this line https://github.com/ucb-bar/gemmini/blob/master/src/main/scala/gemmini/CustomConfigs.scala#L45 for
~~~
val customConfig = defaultFpConfig
~~~

And I rebuilt spike with the script mentioned.
But I am getting the same error.

Do I have to change something on spike or in any other folder?

hngenc, 2022-05-05:
No, that should have been sufficient.

Can you share your `MobileNetV2_0p5_all.onnx` and `images/peon_blanco_fondo_blanco.jpg` files? (And ideally even your `ort_test` binary file).

I'll try to replicate this.

mbelda, 2022-05-05:
Ok.

The model: [MobileNetV2_0p5_all.zip](https://github.com/ucb-bar/onnxruntime-riscv/files/8636064/MobileNetV2_0p5_all.zip)
The image: [peon_blanco](https://user-images.githubusercontent.com/11842513/167026260-23f85dab-4d56-451f-a658-6621f77aaf6c.jpg)
The binary: [ort_test.zip](https://github.com/ucb-bar/onnxruntime-riscv/files/8636077/ort_test.zip)

hngenc, 2022-05-06:
Sorry, it turns out that I gave you incorrect instructions. You should use this config instead:

https://github.com/ucb-bar/gemmini/blob/master/src/main/scala/gemmini/CustomConfigs.scala#L30

(If you want to be safe, you should also set `has_max_pool = true` in that config).

Then, after running `./scripts/build-spike.sh`, your model should work.

This is the output I got from your model:

```
Element count 13. Top 5 classes:
0.000141 r
0.006795 P
0.007563 N
0.016117 n
0.969366 B
```

Let me know if that's what you expected.

mbelda, 2022-05-06:
First, it doesn't work for me and second, it should have predicted P with almost 1.0 probability (at least, that is what the model predicts with onnx standard). I think we provided a model with NHWC config, is that what you expect to get?

I changed to that config with ` has_max_pool = true ` and I keep getting the same error.

I must have something configured the wrong way.
I patched the pk as said in https://github.com/ucb-bar/onnxruntime-riscv/blob/2021-12-23/systolic_runner/docs/BUILD.md.
The actual code needs the patches on different lines, but everything is similar to what is explained.
I attach my `pk/syscalls.h` and `pk/syscalls.c` just in case you need it.
[syscall.h.zip](https://github.com/ucb-bar/onnxruntime-riscv/files/8641145/syscall.h.zip)
[syscall.c.zip](https://github.com/ucb-bar/onnxruntime-riscv/files/8641146/syscall.c.zip)

But when it comes to the RoCC enable, the actual code is very different than the one in the issue https://github.com/riscv/riscv-pk/commit/c53de08b9ba719f3e7b02fc1a029d194a190da48.

I added the lines of the commit when it seemed logic.
I attach my `pk/machine/encoding.h` and `pk/machine/minit.c`
[encoding.h.zip](https://github.com/ucb-bar/onnxruntime-riscv/files/8641156/encoding.h.zip)
[minit.c.zip](https://github.com/ucb-bar/onnxruntime-riscv/files/8641160/minit.c.zip)

I pulled master on `riscv-isa-sim` and now I got the commit bd19c16c779d419c322790b81a5ac57e47773c1c

And I checked that gemini_params.h are the same both on onnx and on spike with
~~~
diff chipyard/generators/gemmini/software/gemmini-rocc-tests/include/gemmini_params.h chipyard/toolchains/esp-tools/riscv-isa-sim/gemmini/gemmini_params.h
~~~

But they are not equal to `systolic_params_fp32.h` Maybe there is the problem.

~~~
diff ./toolchains/esp-tools/riscv-isa-sim/gemmini/gemmini_params.h ./generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/mlas/lib/systolic/systolic_params_fp32.h
~~~

~~~
8c8
< #define DIM 16

> #define DIM 8
12c12
< #define ACC_ROWS 1024

> #define ACC_ROWS 4096
14c14
< #define MAX_BLOCK_LEN (MAX_BYTES/(DIM*1))

> #define MAX_BLOCK_LEN (MAX_BYTES/(DIM*4))
17,21c17,29
< typedef int8_t elem_t;
< static const elem_t elem_t_max = 127;
< static const elem_t elem_t_min = -128;
< typedef int32_t acc_t;
< typedef int64_t full_t;

> typedef float elem_t;
> static const elem_t elem_t_max = 3.4028235E38;
> static const elem_t elem_t_min = -3.4028235E38;
> typedef float acc_t;
> typedef double full_t;
> 
> #define ELEM_T_IS_FLOAT
> #define ELEM_T_EXP_BITS 8
> #define ELEM_T_SIG_BITS 24
> #define ACC_T_EXP_BITS 8
> #define ACC_T_SIG_BITS 24
> typedef uint32_t elem_t_bits;
> typedef uint32_t acc_t_bits;
27c35,36
< typedef int32_t scale_acc_t;

> #define HAS_MVIN_ACC_SCALE
> typedef float scale_acc_t;
40d48
< // Rounding right shift equation: https://riscv.github.io/documents/riscv-v-spec/#_vector_fixed_point_rounding_mode_register_vxrm
42,44c50
<     ((shift) > 0 ? (((x) >> (shift)) + \
<         (((shift) == 0 ? 0 : (((x) >> ((shift)-1)) & 1)) & \
<              ((((shift) <= 1 ? 0 : ((x) & ((1 << ((shift)-1)) - 1))) != 0) | (((x) >> (shift)) & 1)))) : ((x) << (-(shift))))

>     ((x) / (1 << (shift)))
69c75
<     ({float y = ROUND_NEAR_EVEN((x) * (scale)); y > INT8_MAX ? INT8_MAX : (y < INT8_MIN ? INT8_MIN : (acc_t)y);})

>     ((x) * (scale))
72c78
<     ({float y = ROUND_NEAR_EVEN((x) * (scale)); y > INT8_MAX ? INT8_MAX : (y < INT8_MIN ? INT8_MIN : (elem_t)y);})

>     ((x) * (scale))
74c80,81
< #define MVIN_SCALE_ACC(x, scale) (x)

> #define MVIN_SCALE_ACC(x, scale) \
>     ((x) * (scale))
~~~

hngenc, 2022-05-06:
Spike should have `#define DIM 8`. So I suspect that the `scripts/build-spike.sh` command did not succeed.

Can you also share your `GemminiCustomConfigs.scala`? (Perhaps there's a syntax error in there, which prevented `scripts/build-spike.sh` from completing successfully?)

Gemmini does expect the NHWC format, so I'm surprised it didn't work for you...

Could you try running int8 inference on the Mobilenet models we provide here, and seeing if those work for you:

https://github.com/ucb-bar/onnxruntime-riscv/releases/tag/v0.01

mbelda, 2022-05-06:
I have tried with your Mobilenet2.onnx and it gives the same error.

This is the output I receive when executing `scripts/build-spike.sh`
~~~
Generating new gemmini_params.h file...
/home/user/Documents/chipyard/generators/gemmini
Makefile:349: warning: overriding recipe for target 'disasm.o'
Makefile:349: warning: ignoring old recipe for target 'disasm.o'
make: Circular libhwacha.so <- libhwacha.so dependency dropped.
g++ -MMD -MP -I/include/boost-0 -DPREFIX=\"/home/user/Documents/chipyard/esp-tools-install\" -Wall -Wno-unused -g -O2 -fPIC -std=c++11 -g -O2   -I. -I.. -I../fesvr -I../riscv -I../hwacha -I../gemmini -I../sha3 -I../disasm -I../customext -I../fdt -I../softfloat -I../spike_main -I../spike_dasm  -fPIC -c ../gemmini/gemmini.cc
make: Circular libgemmini.so <- libgemmini.so dependency dropped.
g++ -L. -Wl,--export-dynamic -L/lib64  -Wl,-rpath,/home/user/Documents/chipyard/esp-tools-install/lib  -shared -o libgemmini.so  gemmini.o libsoftfloat.a libriscv.a libspike_main.a  libsoftfloat.a  libriscv.a  libspike_main.a -lpthread -ldl -lpthread   
make: Circular libsha3.so <- libsha3.so dependency dropped.
make: Circular libcustomext.so <- libcustomext.so dependency dropped.
make: Circular libsoftfloat.so <- libsoftfloat.so dependency dropped.
Makefile:349: warning: overriding recipe for target 'disasm.o'
Makefile:349: warning: ignoring old recipe for target 'disasm.o'
../scripts/mk-install-dirs.sh /home/user/Documents/chipyard/esp-tools-install/include
for file in fesvr/byteorder.h fesvr/elf.h fesvr/elfloader.h fesvr/htif.h fesvr/dtm.h fesvr/memif.h fesvr/syscall.h fesvr/context.h fesvr/htif_pthread.h fesvr/htif_hexwriter.h fesvr/option_parser.h fesvr/term.h fesvr/device.h fesvr/rfb.h fesvr/tsi.h riscv/mmio_plugin.h; \
  do \
	../scripts/mk-install-dirs.sh /home/user/Documents/chipyard/esp-tools-install/include/`dirname $file`; \
    /usr/bin/install -c -m 644 ../$file /home/user/Documents/chipyard/esp-tools-install/include/`dirname $file`; \
  done
../scripts/mk-install-dirs.sh /home/user/Documents/chipyard/esp-tools-install/include
for dir in fesvr          ; \
  do \
	../scripts/mk-install-dirs.sh /home/user/Documents/chipyard/esp-tools-install/include/$dir; \
    /usr/bin/install -c -m 644 config.h /home/user/Documents/chipyard/esp-tools-install/include/$dir; \
  done
make: Circular libhwacha.so <- libhwacha.so dependency dropped.
make: Circular libgemmini.so <- libgemmini.so dependency dropped.
make: Circular libsha3.so <- libsha3.so dependency dropped.
make: Circular libcustomext.so <- libcustomext.so dependency dropped.
make: Circular libsoftfloat.so <- libsoftfloat.so dependency dropped.
../scripts/mk-install-dirs.sh /home/user/Documents/chipyard/esp-tools-install/lib
for file in libfesvr.a libhwacha.so libgemmini.so libsha3.so libdisasm.a libcustomext.so libsoftfloat.so; \
  do \
    /usr/bin/install -c -m 644 $file /home/user/Documents/chipyard/esp-tools-install/lib; \
  done
../scripts/mk-install-dirs.sh /home/user/Documents/chipyard/esp-tools-install/bin
for file in elf2hex spike spike-log-parser xspike termios-xspike spike-dasm; \
  do \
    /usr/bin/install -c -m 755 $file /home/user/Documents/chipyard/esp-tools-install/bin; \
  done
../scripts/mk-install-dirs.sh /home/user/Documents/chipyard/esp-tools-install/lib/pkgconfig/
for file in riscv-fesvr.pc riscv-disasm.pc; \
  do \
    /usr/bin/install -c -m 644 $file /home/user/Documents/chipyard/esp-tools-install/lib/pkgconfig/; \
  done
~~~

I attach my `gemmini/CustomConfig.scala`
[CustomConfigs.scala.zip](https://github.com/ucb-bar/onnxruntime-riscv/files/8643138/CustomConfigs.scala.zip)

hngenc, 2022-05-06:
After running `./scripts/build-spike.sh`, there should be a `build.log` file in `chipyard/sims/verilator/`. Can you also attach that file?

hngenc, 2022-05-06:
Also, one other thing:

Can you make sure that your `chipyard/generators/gemmini/src/main/scala/gemmini/CustomConfigs.scala` file is exactly the same as your `chipyard/generators/gemmini/configs/GemminiCustomConfigs.scala` file?

They should just be softlinks to each other, but I've seen them get out-of-sync before. I just want to be sure that didn't happen to you.

mbelda, 2022-05-06:
Here it goes the build.log
[build.log.zip](https://github.com/ucb-bar/onnxruntime-riscv/files/8643467/build.log.zip)
 
I don't have any configs folder at `chipyard/generators/gemmini`, I just have this:
~~~
build.sbt  CHIPYARD.hash  img  LICENSE  project  README.md  scalastyle-config.xml  scalastyle-test-config.xml  scripts  software  SPIKE.hash  src  target
~~~
I searched for `GemminiCustomConfigs.scala` and I don't have it either on any subfolder of gemmini.

hngenc, 2022-05-06:
Oh, you need to run `scripts/setup-paths.scala`, and that will create the config files for you.

Can you run that and make sure that your `configs/GemminiCustomConfigs.scala` file is using the `trainingConfig`?

I'll check your `build.log` and get back to you.

mbelda, 2022-05-06:
I executed that script and rebuild the repo and now it works and I obtain the same inference from -x 0 (CPU) and -x 2 (WS).
So it works fine.
I must have missed that script at some point.

Thank you very much!

hngenc, 2022-05-06:
No worries; glad it worked!

closed 2022-05-13 by mbelda

aladerran, 2023-08-15:
@hngenc Hi guys, a quick question:
I am also trying to run a fp32 model using imagenet runner. I followed the same procedure as @mbelda did and I can successfully run the fp32 model.
One slight difference that confuses me is that after I run ```./scripts/build-spike.sh```, spike ```#define DIM 4``` in the gemmini_params.h, which is neither 16 nor 8 you mentioned above.

```
#ifndef GEMMINI_PARAMS_H
#define GEMMINI_PARAMS_H

#include <stdint.h>
#include <limits.h>

#define XCUSTOM_ACC 3
#define DIM 4
#define ADDR_LEN 32
#define BANK_NUM 4
#define BANK_ROWS 4096
#define ACC_ROWS 4096
#define MAX_BYTES 64
#define MAX_BLOCK_LEN (MAX_BYTES/(DIM*4))
#define MAX_BLOCK_LEN_ACC (MAX_BYTES/(DIM*4))
```

Should the DIM be 4 or I am missing some steps?

ethan2339, 2025-03-11:
Test comment to retrieve issue comments.