ucb-bar/onnxruntime-riscv #119: Failing to run the calibrate

HJL8055, 2023-03-13:
Hello,
I am trying to run an int8-quantized model with clip node. But after I run the calibrate.py by the code below, it seems that I get a brken quantized model.
`python calibrate.py --model_path V4.onnx  --dataset_path ../../pb --output_model_path model_quantized.onnx --static=True --data_preprocess=None --mode=int8` 
When I try to execute the systolic_runner example minimaly modified on spike, I keep getting the following error: 
[image]
Here comes the netron view of the brken quantized model. It seems that I lose all of my clip nodes.
[image]

hngenc, 2023-11-27:
Can you share your model? I can try to replicate this error

closed 2023-12-01 by HJL8055

HJL8055, 2023-12-01:
> Can you share your model? I can try to replicate this error

Thanks a lot for your reply but I have already solved this error. The operator——Relu6 was used in my model, which was regarded as Relu+clip. However, the quantize tool couldn't handle nodes of clip correctly, causing the broken nodes eventually.

hngenc, 2023-12-01:
Thanks for the update!