ucb-bar/onnxruntime-riscv #121: ORT building error with newest gemmini

aladerran, 2023-07-12:
**Describe the bug**
ORT building error with gemmini v0.7.1

**Urgency**
Hope to get your answer asap, thanks!

**System information**
- OS Platform and Distribution (e.g., Linux Ubuntu 16.04): CentOS Linux release 7.9.2009 (Core)
- ONNX Runtime installed from (source or binary):
- ONNX Runtime version: 1.9.0
- Python version: 3.10.10
- Visual Studio version (if applicable):
- GCC/Compiler version (if compiling from source): 8.2.0
- CUDA/cuDNN version: 11.6
- GPU model and memory:

**To Reproduce**
After building gemmini v0.7.1 and chipyard 1.9.0 and submoudule the lastest version of onnx-runtime-riscv, running ./build.sh --config=Release --parallel --enable_training or ./build.sh --config=Release --parallel under onnx-runtime-riscv.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
(/ugra/srlin/chipyard/.conda-env) srlin@chensvr:~/chipyard/generators/gemmini/software/onnxruntime-riscv$ ./build.sh --config=Release --parallel --enable_training
Performing Release build
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 1602k  100 1602k    0     0   675k      0  0:00:02  0:00:02 --:--:--  882k
Archive:  protoc.zip
   creating: include/
   creating: include/google/
   creating: include/google/protobuf/
  inflating: include/google/protobuf/wrappers.proto  
  inflating: include/google/protobuf/field_mask.proto  
  inflating: include/google/protobuf/api.proto  
  inflating: include/google/protobuf/struct.proto  
  inflating: include/google/protobuf/descriptor.proto  
  inflating: include/google/protobuf/timestamp.proto  
   creating: include/google/protobuf/compiler/
  inflating: include/google/protobuf/compiler/plugin.proto  
  inflating: include/google/protobuf/empty.proto  
  inflating: include/google/protobuf/any.proto  
  inflating: include/google/protobuf/source_context.proto  
  inflating: include/google/protobuf/type.proto  
  inflating: include/google/protobuf/duration.proto  
   creating: bin/
  inflating: bin/protoc              
  inflating: readme.txt              
2023-07-12 22:33:44,288 tools_python_utils [INFO] - flatbuffers module is not installed. parse_config will not be available
2023-07-12 22:33:44,292 build [INFO] - Build started
2023-07-12 22:33:44,292 build [INFO] - Generating CMake build tree
2023-07-12 22:33:44,293 util.run [INFO] - Running subprocess in 'build/Release'
['/ugra/srlin/chipyard/.conda-env/bin/cmake', '/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/cmake', '-DBUILD_SHARED_LIBS=OFF', '-DCMAKE_EXE_LINKER_FLAGS=-latomic -static', '-DCMAKE_SYSTEM_NAME=Linux', '-DUNIX=True', '-Donnxruntime_USE_SYSTOLIC=ON', '-Donnxruntime_FOR_FIRESIM=OFF', '-Donnxruntime_USE_HWACHA=OFF', '-Donnxruntime_RUN_ONNX_TESTS=OFF', '-Donnxruntime_BUILD_WINML_TESTS=ON', '-Donnxruntime_GENERATE_TEST_REPORTS=ON', '-Donnxruntime_DEV_MODE=ON', '-DPYTHON_EXECUTABLE=/ugra/srlin/chipyard/.conda-env/bin/python3', '-Donnxruntime_USE_CUDA=OFF', '-Donnxruntime_CUDA_VERSION=', '-Donnxruntime_ROCM_VERSION=', '-Donnxruntime_CUDA_HOME=', '-Donnxruntime_CUDNN_HOME=', '-Donnxruntime_USE_FEATURIZERS=OFF', '-Donnxruntime_USE_MIMALLOC_STL_ALLOCATOR=OFF', '-Donnxruntime_USE_MIMALLOC_ARENA_ALLOCATOR=OFF', '-Donnxruntime_ENABLE_PYTHON=OFF', '-Donnxruntime_BUILD_CSHARP=OFF', '-Donnxruntime_BUILD_JAVA=OFF', '-Donnxruntime_BUILD_NODEJS=OFF', '-Donnxruntime_BUILD_OBJC=OFF', '-Donnxruntime_BUILD_SHARED_LIB=OFF', '-Donnxruntime_BUILD_APPLE_FRAMEWORK=OFF', '-Donnxruntime_USE_DNNL=OFF', '-Donnxruntime_DNNL_GPU_RUNTIME=', '-Donnxruntime_DNNL_OPENCL_ROOT=', '-Donnxruntime_USE_NNAPI_BUILTIN=OFF', '-Donnxruntime_USE_RKNPU=OFF', '-Donnxruntime_USE_OPENMP=OFF', '-Donnxruntime_USE_TVM=OFF', '-Donnxruntime_USE_LLVM=OFF', '-Donnxruntime_ENABLE_MICROSOFT_INTERNAL=OFF', '-Donnxruntime_USE_VITISAI=OFF', '-Donnxruntime_USE_NUPHAR=OFF', '-Donnxruntime_USE_TENSORRT=OFF', '-Donnxruntime_TENSORRT_HOME=', '-Donnxruntime_USE_MIGRAPHX=OFF', '-Donnxruntime_MIGRAPHX_HOME=', '-Donnxruntime_CROSS_COMPILING=ON', '-Donnxruntime_DISABLE_CONTRIB_OPS=OFF', '-Donnxruntime_DISABLE_ML_OPS=OFF', '-Donnxruntime_DISABLE_RTTI=OFF', '-Donnxruntime_DISABLE_EXCEPTIONS=OFF', '-Donnxruntime_DISABLE_ORT_FORMAT_LOAD=OFF', '-Donnxruntime_MINIMAL_BUILD=OFF', '-Donnxruntime_EXTENDED_MINIMAL_BUILD=OFF', '-Donnxruntime_MINIMAL_BUILD_CUSTOM_OPS=OFF', '-Donnxruntime_REDUCED_OPS_BUILD=OFF', '-Donnxruntime_MSVC_STATIC_RUNTIME=OFF', '-Donnxruntime_ENABLE_LANGUAGE_INTEROP_OPS=OFF', '-Donnxruntime_USE_DML=OFF', '-Donnxruntime_USE_WINML=OFF', '-Donnxruntime_BUILD_MS_EXPERIMENTAL_OPS=OFF', '-Donnxruntime_USE_TELEMETRY=OFF', '-Donnxruntime_ENABLE_LTO=OFF', '-Donnxruntime_USE_ACL=OFF', '-Donnxruntime_USE_ACL_1902=OFF', '-Donnxruntime_USE_ACL_1905=OFF', '-Donnxruntime_USE_ACL_1908=OFF', '-Donnxruntime_USE_ACL_2002=OFF', '-Donnxruntime_USE_ARMNN=OFF', '-Donnxruntime_ARMNN_RELU_USE_CPU=ON', '-Donnxruntime_ARMNN_BN_USE_CPU=ON', '-Donnxruntime_ENABLE_NVTX_PROFILE=OFF', '-Donnxruntime_ENABLE_TRAINING=ON', '-Donnxruntime_ENABLE_TRAINING_OPS=OFF', '-Donnxruntime_ENABLE_CPU_FP16_OPS=ON', '-Donnxruntime_USE_NCCL=ON', '-Donnxruntime_BUILD_BENCHMARKS=OFF', '-Donnxruntime_USE_ROCM=OFF', '-Donnxruntime_ROCM_HOME=', '-DOnnxruntime_GCOV_COVERAGE=OFF', '-Donnxruntime_USE_MPI=ON', '-Donnxruntime_ENABLE_MEMORY_PROFILE=OFF', '-Donnxruntime_ENABLE_CUDA_LINE_NUMBER_INFO=OFF', '-Donnxruntime_BUILD_WEBASSEMBLY=OFF', '-Donnxruntime_ENABLE_WEBASSEMBLY_EXCEPTION_CATCHING=ON', '-Donnxruntime_ENABLE_WEBASSEMBLY_THREADS=OFF', '-Donnxruntime_ENABLE_EAGER_MODE=OFF', '-Donnxruntime_ENABLE_WEBASSEMBLY_SOURCEMAP=OFF', '-Donnxruntime_WEBASSEMBLY_SOURCEMAP_BASE=', '-DCMAKE_SYSTEM_PROCESSOR=riscv', '-DONNX_CUSTOM_PROTOC_EXECUTABLE=/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/build/protoc/bin/protoc', '-Donnxruntime_PYBIND_EXPORT_OPSCHEMA=OFF', '-Donnxruntime_ENABLE_MEMLEAK_CHECKER=OFF', '-DCMAKE_BUILD_TYPE=Release']
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /ugra/srlin/chipyard/.conda-env/riscv-tools/bin/riscv64-unknown-linux-gnu-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /ugra/srlin/chipyard/.conda-env/riscv-tools/bin/riscv64-unknown-linux-gnu-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test onnxruntime_HAVE_BUILTIN_ATOMICS
[... 195 lines cut ...]
[ 48%] Built target onnxruntime_framework
gmake[1]: *** [CMakeFiles/onnxruntime_graph.dir/all] Error 2
[ 48%] Linking CXX static library libonnxruntime_optimizer.a
[ 48%] Built target onnxruntime_optimizer
gmake: *** [all] Error 2
Traceback (most recent call last):
  File "/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/tools/ci_build/build.py", line 2162, in <module>
    sys.exit(main())
  File "/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/tools/ci_build/build.py", line 2088, in main
    build_targets(args, cmake_path, build_dir, configs, num_parallel_jobs, args.target)
  File "/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/tools/ci_build/build.py", line 1087, in build_targets
    run_subprocess(cmd_args, env=env)
  File "/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/tools/ci_build/build.py", line 591, in run_subprocess
    return run(*args, cwd=cwd, capture_stdout=capture_stdout, shell=shell, env=my_env)
  File "/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/tools/python/util/run.py", line 41, in run
    completed_process = subprocess.run(
  File "/ugra/srlin/chipyard/.conda-env/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/ugra/srlin/chipyard/.conda-env/bin/cmake', '--build', 'build/Release', '--config', 'Release', '--', '-j80']' returned non-zero exit status 2.

The above error occurs when I turned FP32 on and Int8 off as well.

I have tried the same procedure in my ubuntu20.04 virtual machine with older version of gemmini which uses esp-tools and the result was fine. I am wondering if it could be a new issue caused by newer version of gemmini. 

Thanks for your help!

aladerran, 2023-07-20:
Hi. I found that when building without "--parallel" option, it triggers: 

```
[ 51%] Built target onnxruntime_optimizer
[ 51%] Building CXX object CMakeFiles/onnxruntime_session.dir/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/session/IOBinding.cc.o
[ 51%] Building CXX object CMakeFiles/onnxruntime_session.dir/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/session/abi_session_options.cc.o
[ 51%] Building CXX object CMakeFiles/onnxruntime_session.dir/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/session/custom_ops.cc.o
[ 51%] Building CXX object CMakeFiles/onnxruntime_session.dir/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/session/default_cpu_allocator_c_api.cc.o
[ 51%] Building CXX object CMakeFiles/onnxruntime_session.dir/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/session/device_allocator.cc.o
[ 51%] Building CXX object CMakeFiles/onnxruntime_session.dir/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/session/environment.cc.o
[ 51%] Building CXX object CMakeFiles/onnxruntime_session.dir/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/session/inference_session.cc.o
In file included from /ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/session/inference_session_utils.h:21,
                 from /ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/session/inference_session.cc:50:
/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/cmake/external/json/single_include/nlohmann/json.hpp: In instantiation of 'std::string nlohmann::detail::lexer<BasicJsonType>::get_token_string() const [with BasicJsonType = nlohmann::basic_json<>; std::string = std::__cxx11::basic_string<char>]':
/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/cmake/external/json/single_include/nlohmann/json.hpp:8745:57:   required from 'void nlohmann::detail::parser<BasicJsonType>::parse(bool, BasicJsonType&) [with BasicJsonType = nlohmann::basic_json<>]'
/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/cmake/external/json/single_include/nlohmann/json.hpp:20835:79:   required from 'static nlohmann::basic_json<ObjectType, ArrayType, StringType, BooleanType, NumberIntegerType, NumberUnsignedType, NumberFloatType, AllocatorType, JSONSerializer> nlohmann::basic_json<ObjectType, ArrayType, StringType, BooleanType, NumberIntegerType, NumberUnsignedType, NumberFloatType, AllocatorType, JSONSerializer>::parse(IteratorType, IteratorType, parser_callback_t, bool) [with IteratorType = const char*; typename std::enable_if<std::is_base_of<std::random_access_iterator_tag, typename std::iterator_traits<_InputIterator>::iterator_category>::value, int>::type <anonymous> = 0; ObjectType = std::map; ArrayType = std::vector; StringType = std::__cxx11::basic_string<char>; BooleanType = bool; NumberIntegerType = long int; NumberUnsignedType = long unsigned int; NumberFloatType = double; AllocatorType = std::allocator; JSONSerializer = nlohmann::adl_serializer; parser_callback_t = std::function<bool(int, nlohmann::detail::parser<nlohmann::basic_json<> >::parse_event_t, nlohmann::basic_json<>&)>]'
/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/cmake/external/json/single_include/nlohmann/json.hpp:22653:33:   required from here
/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/cmake/external/json/single_include/nlohmann/json.hpp:8494:24: error: comparison is always true due to limited range of data type [-Werror=type-limits]
 8494 |             if ('\x00' <= c and c <= '\x1F')
      |                 ~~~~~~~^~~~
cc1plus: all warnings being treated as errors
gmake[2]: *** [CMakeFiles/onnxruntime_session.dir/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/onnxruntime/core/session/inference_session.cc.o] Error 1
gmake[1]: *** [CMakeFiles/onnxruntime_session.dir/all] Error 2
gmake: *** [all] Error 2
Traceback (most recent call last):
  File "/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/tools/ci_build/build.py", line 2162, in <module>
    sys.exit(main())
  File "/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/tools/ci_build/build.py", line 2088, in main
    build_targets(args, cmake_path, build_dir, configs, num_parallel_jobs, args.target)
  File "/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/tools/ci_build/build.py", line 1087, in build_targets
    run_subprocess(cmd_args, env=env)
  File "/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/tools/ci_build/build.py", line 591, in run_subprocess
    return run(*args, cwd=cwd, capture_stdout=capture_stdout, shell=shell, env=my_env)
  File "/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/tools/python/util/run.py", line 41, in run
    completed_process = subprocess.run(
  File "/ugra/srlin/chipyard/.conda-env/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/ugra/srlin/chipyard/.conda-env/bin/cmake', '--build', 'build/Release', '--config', 'Release']' returned non-zero exit status 2.
```

So I modified the "auto c" to "signed char c"  in the following snippet:

```
[... cut ...]
  File "/ugra/srlin/chipyard/generators/gemmini/software/onnxruntime-riscv/tools/python/util/run.py", line 41, in run
    completed_process = subprocess.run(
  File "/ugra/srlin/chipyard/.conda-env/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/ugra/srlin/chipyard/.conda-env/bin/cmake', '--build', 'build/Release', '--config', 'Release', '--', '-j80']' returned non-zero exit status 2.
```

This one I can't handle = =

aladerran, 2023-07-23:
I also want to make some changes for systolic.h,and I am now troubled by the "-Werror" CXX flag. Would you please tell me how to unset such flag while building?

closed 2023-11-11 by aladerran

aladerran, 2023-12-12:
@8u9mak3r May help if you are using risc-v toolchains: comment out all the -Werror in the CMake file.

8u9mak3r, 2023-12-14:
@aladerran Wow, it worked for me! Thanks 4 your help!