ucb-bar/onnxruntime-riscv #124: error:./scripts/build-onnx-inference.sh

GGKOP, 2024-01-22:
I want to run ONNX Model On Spike: Inference

I clone the code
`https://github.com/ucb-bar/onnxruntime-riscv.git`
then
`git submodule update --init --recursive`
`./scripts/build-onnx-inference.sh`

```
 15%] Linking CXX static library libtensorboard.a
[ 15%] Built target tensorboard
gmake: *** [Makefile:166: all] Error 2
Traceback (most recent call last):
  File "/home/ricky/chipyard/generators/gemmini/software/onnxruntime-riscv/tools/ci_build/build.py", line 2162, in <module>
    sys.exit(main())
  File "/home/ricky/chipyard/generators/gemmini/software/onnxruntime-riscv/tools/ci_build/build.py", line 2088, in main
    build_targets(args, cmake_path, build_dir, configs, num_parallel_jobs, args.target)
  File "/home/ricky/chipyard/generators/gemmini/software/onnxruntime-riscv/tools/ci_build/build.py", line 1087, in build_targets
    run_subprocess(cmd_args, env=env)
  File "/home/ricky/chipyard/generators/gemmini/software/onnxruntime-riscv/tools/ci_build/build.py", line 591, in run_subprocess
    return run(*args, cwd=cwd, capture_stdout=capture_stdout, shell=shell, env=my_env)
  File "/home/ricky/chipyard/generators/gemmini/software/onnxruntime-riscv/tools/python/util/run.py", line 41, in run
    completed_process = subprocess.run(
  File "/home/ricky/chipyard/.conda-env/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/home/ricky/chipyard/.conda-env/bin/cmake', '--build', 'build/Debug', '--config', 'Debug', '--', '-j32']' returned non-zero exit status 2.
Building against Debug
/home/ricky/chipyard/.conda-env/riscv-tools/bin/../lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find ../..//build/Debug/libonnx_test_runner_common.a: No such file or directory
/home/ricky/chipyard/.conda-env/riscv-tools/bin/../lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find ../..//build/Debug/libonnxruntime_test_utils.a: No such file or directory
/home/ricky/chipyard/.conda-env/riscv-tools/bin/../lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find ../..//build/Debug/libonnxruntime_training_runner.a: No such file or directory
/home/ricky/chipyard/.conda-env/riscv-tools/bin/../lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find ../..//build/Debug/libonnxruntime_training.a: No such file or directory
/home/ricky/chipyard/.conda-env/riscv-tools/bin/../lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find ../..//build/Debug/libonnxruntime_session.a: No such file or directory
/home/ricky/chipyard/.conda-env/riscv-tools/bin/../lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find ../..//build/Debug/libonnxruntime_optimizer.a: No such file or directory
/home/ricky/chipyard/.conda-env/riscv-tools/bin/../lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find ../..//build/Debug/libonnxruntime_providers.a: No such file or directory
/home/ricky/chipyard/.conda-env/riscv-tools/bin/../lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find ../..//build/Debug/libonnxruntime_util.a: No such file or directory
/home/ricky/chipyard/.conda-env/riscv-tools/bin/../lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find ../..//build/Debug/libonnxruntime_framework.a: No such file or directory
/home/ricky/chipyard/.conda-env/riscv-tools/bin/../lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find ../..//build/Debug/libonnxruntime_util.a: No such file or directory
/home/ricky/chipyard/.conda-env/riscv-tools/bin/../lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find ../..//build/Debug/libonnxruntime_graph.a: No such file or directory
/home/ricky/chipyard/.conda-env/riscv-tools/bin/../lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find ../..//build/Debug/libonnxruntime_providers_systolic.a: No such file or directory
/home/ricky/chipyard/.conda-env/riscv-tools/bin/../lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find ../..//build/Debug/libonnxruntime_common.a: No such file or directory
/home/ricky/chipyard/.conda-env/riscv-tools/bin/../lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find ../..//build/Debug/libonnxruntime_flatbuffers.a: No such file or directory
/home/ricky/chipyard/.conda-env/riscv-tools/bin/../lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find ../..//build/Debug/libonnx_test_data_proto.a: No such file or directory
/home/ricky/chipyard/.conda-env/riscv-tools/bin/../lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find ../..//build/Debug/external/onnx/libonnx.a: No such file or directory
/home/ricky/chipyard/.conda-env/riscv-tools/bin/../lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find ../..//build/Debug/external/onnx/libonnx_proto.a: No such file or directory
collect2: error: ld returned 1 exit status
make: *** [Makefile:11: ort_test] Error 1
Please ignore any dlopen warning above. Glibc hates being statically linked.

```

add info
```

 CMake version             : 3.26.3
--   CMake command             : /home/ricky/chipyard/.conda-env/bin/cmake
--   System                    : Linux
--   C++ compiler              : /home/ricky/chipyard/.conda-env/riscv-tools/bin/riscv64-unknown-linux-gnu-g++
--   C++ compiler version      : 12.2.0
--   CXX flags                 : -march=rv64imafdc -mabi=lp64d -ffunction-sections -fdata-sections -Wno-error=attributes -Wnon-virtual-dtor
--   Build type                : Debug
--   Compile definitions       : ENABLE_ORT_FORMAT_LOAD;EIGEN_MPL2_ONLY;PLATFORM_POSIX
--   CMAKE_PREFIX_PATH         : 
--   CMAKE_INSTALL_PREFIX      : /usr/local
--   CMAKE_MODULE_PATH         : /home/ricky/chipyard/generators/gemmini/software/onnxruntime-riscv/cmake/external
-- 
--   ONNX version              : 1.9.0
--   ONNX NAMESPACE            : onnx
--   ONNX_USE_LITE_PROTO       : ON
--   USE_PROTOBUF_SHARED_LIBS  : OFF
--   ONNX_DISABLE_EXCEPTIONS   : OFF
--   ONNX_WERROR               : OFF
--   ONNX_BUILD_TESTS          : OFF
--   ONNX_BUILD_BENCHMARKS     : OFF
--   ONNXIFI_DUMMY_BACKEND     : OFF
--   ONNXIFI_ENABLE_EXT        : OFF
-- 
--   Protobuf compiler         : 
--   Protobuf includes         : 
--   Protobuf libraries        : 
--   BUILD_ONNX_PYTHON         : OFF

```
  what can I do？

vvcvv-as, 2024-05-28:
Same problem here. Did you find the solution?

hansa31, 2024-12-05:
@hngenc Can you please look into this issue?

ethan2339, 2025-03-09:
Same problem here. I need the guide please..